arendst/Sonoff-MQTT-OTA-Arduino #75: Compiling problem

scargill, 2016-11-12:
Hi

I thought I'd give your software a go... beats re-inventing the wheel for the POW!

However, I had Arduino 1,6,9 - trouble compiling - upgraded to 1.6.12 - same problem - two issues - I wonder if you can help (got the latest pubsubclient - compiled for generic esp8266  and Spiffs as per your notes and.. any ideas??

Build options changed, rebuilding all
C:\Users\Peter\Dropbox\home control\ESP\Sonoff-MQTT-OTA-Arduino\sonoff\sonoff.ino: In function 'void stateloop()':

sonoff:1656: error: 'class ESP8266HTTPUpdate' has no member named 'rebootOnUpdate'

         ESPhttpUpdate.rebootOnUpdate(false);

                       ^

C:\Users\Peter\Dropbox\home control\ESP\Sonoff-MQTT-OTA-Arduino\sonoff\sonoff.ino: In function 'void setup()':

sonoff:1799: error: 'class HardwareSerial' has no member named 'baudRate'

   if (Serial.baudRate() != Baudrate) {

              ^

C:\Users\Peter\Dropbox\home control\ESP\Sonoff-MQTT-OTA-Arduino\sonoff\webserver.ino: In function 'void handleInfo()':

webserver:830: error: 'class EspClass' has no member named 'getCoreVersion'

   page += F("<tr><td><b>Core/SDK version</b></td><td>"); page += ESP.getCoreVersion(); page += F("/"); page += String(ESP.getSdkVersion()); page += F("</td></tr>");

                                                                      ^

exit status 1
'class ESP8266HTTPUpdate' has no member named 'rebootOnUpdate'

arendst, 2016-11-12:
Hi Peter! Nice meeting you again.

I guess the problem is with the esp arduino software. It should be 2.3.0 as that started to support Serial.baudRate and probably the others too.

scargill, 2016-11-12:
And you!

I grabbed the latest ESP ARduino - and the error has changed.. got two now - different ones.

In file included from C:\Users\Peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WiFi\src\WiFiUdp.cpp:41:0:

C:\Users\Peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WiFi\src\include/UdpContext.h: In member function 'void UdpContext::setMulticastInterface(ip_addr_t)':

C:\Users\Peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WiFi\src\include/UdpContext.h:109:48: error: 'udp_set_multicast_netif_addr' was not declared in this scope

         udp_set_multicast_netif_addr(_pcb, addr);

                                                ^

C:\Users\Peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WiFi\src\include/UdpContext.h: In member function 'void UdpContext::setMulticastTTL(int)':

C:\Users\Peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WiFi\src\include/UdpContext.h:117:40: error: 'udp_set_multicast_ttl' was not declared in this scope

         udp_set_multicast_ttl(_pcb, ttl);

                                        ^

exit status 1
Error compiling for board Generic ESP8266 Module.

scargill, 2016-11-12:
Erm 2.30 - I just download 2.1.0 by the look of it - where did you get 2.3.0 from ?

arendst, 2016-11-12:
From https://github.com/esp8266/Arduino or use  http://arduino.esp8266.com/stable/package_esp8266com_index.json into Additional Board Manager URLs field in the Arduino IDE.

scargill, 2016-11-12:
Forget that - using the wrong one - compiled ok - now to actually TRY it.... :-)

scargill, 2016-11-12:
Ok... TH16 and POW programmed.... something I should know about the DUAL - it has 2 connectors neither of which seem to want to program.... anything you're aware of?

arendst, 2016-11-12:
Pete see my wiki ([https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/wiki/Compile-and-Upload](https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/wiki/Compile-and-Upload)) where to connect to and how to put it in update mode (tricky).

scargill, 2016-11-12:
Got it - WHY would they do that.... oh well, all 3 programmed - just have to try it out now - thank you!

arendst, 2016-11-12:
Please be aware that the AM2301 doesn't work just now. I'm still waiting for a sensor to test...

scargill, 2016-11-12:
I use the Dallas and DHT22 pretty much exclusively.

scargill, 2016-11-12:
Interested to know why the mqtt has to be set in the H file ? Any chance of setting in serially? ie without recompiling?

arendst, 2016-11-12:
Yep! there are a host of commands you can use to configure the software. Again see the wiki - right menu: Commands.

arendst, 2016-11-12:
It's ```mqtthost```

arendst, 2016-11-12:
Dallas and DHT22 are supported OK.

Easy config for important items can be done using the web interface too.

scargill, 2016-11-12:
Serial.... not quite following... All I can get back is "ok"...   So if I wanted to set the MQTT to 192.168.0.20 - what exactly would I send to the serial port (at 19,200 followed by carriage return)?

arendst, 2016-11-12:
Again it's in the wiki. For all but dual it's 115200 and Both NL&CR, for dual it's 19200. Why you receive ok is a mistery to me as that string is not in my code...

More info will be shown using command ```seriallog 3```.

arendst, 2016-11-12:
To set MQTT host you would type ```mqtthost 192.168.0.20```

arendst, 2016-11-12:
Just a return should respond with the time

scargill, 2016-11-12:
It does - but how do you send a command via serial? 
Like setting up MQTT? or WIFI?

scargill, 2016-11-12:
if I send mttthost 192.168.0.20 -  what comes back is the command I sent and the time - no confirmation or anything.if I make a spelling mistake - the same thing comes back.

arendst, 2016-11-12:
Well if it comes back it means it received it. Just enable more info by using command ```seriallog 2``` once.

Entering mqtthost without an options returns current value.

scargill, 2016-11-12:
ok, start to get this - got the connection... but the MQTT connection is not having is.

put in mqttuser  mqtthost mqttport mqttpassword -  and "attempting connection" over and over.

if I type the likes of mqttport on it's own -  according to the instructions the port number should come back - I get the time and the word mqtt port - and that's it.

arendst, 2016-11-12:
Me too.... That's because the answer could not be send as it depends on MQTT. Minor problem on my side.

Well at least you have a wifi connection so you could now try to connect to the webserver using the IP it received and showed during the reboot. There you can try to get MQTT going.

scargill, 2016-11-12:
ok - getting there, got WIFI and got MQTT.... now need to get this thing off the tiny power the FTDI can give and onto mails. Will let you know - looking good. I have my own code but as you've down the POW stuff it hardly seems worth the effort if your code works :-)

arendst, 2016-11-12:
Just let me know your experiences. I probably notice in your blog ;-)

arendst, 2016-11-12:
BTW. did you notice the Pow has a 4MByte flash while the others have just 1MByte flashes?

scargill, 2016-11-12:
one last one - and thanks for that 4M tip..

Ok, all setup on serial - it is called sonoff-3054

It isn't incredibly obvious for the notes - what I should send to and what I should subscribe to - in your notes there does not seem to be reference to a particular device.

An example would help.

To turn the first relay on - what would be my TOPIC - and PAYLOAD

And to receive stuff - what would be the TOPIC?

scargill, 2016-11-12:
I've been at them to fit 4 meg flash for AGES - maybe it is rubbing off - I will ask.

arendst, 2016-11-12:
form the user_config.h you might conclude that:

Commands are send to ```cmnd/sonoff-3054/power``` with payload ```on``` or ```1``` or ```off``` or ```0``` or ```toggle``` or ```2```.
Commands can also be send addressing the relay as ```cmnd/sonoff-3054/1/power```.

Receive stuff would be ```stat/sonoff-3054/POWER``` or ```stat/sonoff-3054/1/POWER``` depending on what you used as command. Both button and webpage button always return ```stat/sonoff-3054/1/POWER``` or for the dual also ```stat/sonoff-3054/2/POWER```

You might also expect telemetric stuff from ```tele/sonoff-3054/UPTIME``` ig you enabled it by entering a time interval in seconds with command ```TELEPERIOD```.

Another 'feature' is that both command ```power``` and ```light``` behave the same; once you use one of them all responses will also use the last used name. This was implemented for Domoticz compatibility.

scargill, 2016-11-12:
Ok so I called the device pete - and that's fine turned a light on...

What is HOST??

And if I want to see the state of power

What would be the TOPIC and PAYLOAD and what should be the subscription to get that value back? Assuming user pete.

arendst, 2016-11-12:
State of power? Just enter ```cmnd/pete/POWER``` with no payload or a question mark as payload. It returns a string as defined in user_config.h and is default 'ON' or 'OFF'.

Subscribe to ```stat/pete/#```

scargill, 2016-11-12:
Sorry -got most of it   I'm seeing tele/pete/POWER

But if I send topic cmnd/pete/power with any payload - I don't get that status message back

(just got to get my head around this and all will be well)

arendst, 2016-11-12:
```tele/pete/POWER``` is the result of setting ```TELEPERIOD```. It defaults to 300 seconds and sends status info just for fun or logging.

Only ```cmnd/...``` respond with ```stat/...```.

scargill, 2016-11-12:
Got your message - when I send cmnd/pete/POWER 

Nothing immediately comes back - eventually - maybe a minute later I get 4 items including the state of power - but my "publish" above - does not trigger anything coming back un /tele/pete/#  (I'm using mqtt-spy to test this.   The light is off - as I turned it of.. and a minute later I'm still not seeing anything coming back as I write this. Clearly I'm missing the point -

patbeirne, 2016-11-12:
Pete,

Try subscirbing to `stat/pete/+` or `stat/pete/#` to get everything, just until things stabilize. Then you can subscribe to `stat/pete/POWER` (note the upper case on all `stat`)

arendst, 2016-11-12:
Yes you do. I'm using mqtt-spy too. Just select a default subscribe to #. This should show all messages received.

scargill, 2016-11-12:
AH!!!!
Got it... there's always that lightbulb moment.

scargill, 2016-11-12:
right now to actually go and try stuff out!!!

scargill, 2016-11-12:
Suggestion for you - if Arduino environment will let you do it - I have it but I wrote mine using SDK.. a second WIFI - in case of failure...

So if I pick on one of my boxes in the office and take it into the house - it will try to connect -realise that it is not getting anywhere - and try the SECOND SSID and pass.  Just a thought - right -I'm off to play. Thanks for rapid feedback.

arendst, 2016-11-12:
Ah, I guess I'm never finished...

scargill, 2016-11-12:
I love the web config... mind you, you should say REALLY short presses...  I note you have the Sonoff smart socket but no sign of their new wall mounted WIFI switch - might be nice to add that.

Finished? Never - been working on our code for a year! Sadly it is now too big to fit into the smaller FLASH - let's hope this new trend by ITEAD is something they keep up.

scargill, 2016-11-12:
Erm, on mains... your web interface - two buttons at the top ON and OFF...

They do it twice... so press for on and I can hear ON/OFF/ON -  press  for off and I can hear OFF/ON/OFF -  the MQTT commands work fine..

Any reason for that?

arendst, 2016-11-12:
Yes and No. If it's a dual it should have two buttons but one for each relay. If it's another sonoff it should have one button but the software failed to recognized it's NOT a dual.

Simple solution: Execute undocumented command ```model 1``` once and you're done.

scargill, 2016-11-12:
no that made it a normal TH16 - with one button - this has 2 buttons - it's a dual - and the two buttons are still doing the toggling.

scargill, 2016-11-12:
(I used "model 2" to get it back to dual - and in the web interface the two buttons re-appeared. but they're still doing on-off-on and off-on-off for on or off.

one button press
18:15:45 MQTT: stat/pete/2/POWER = ON
18:15:45 MQTT: stat/pete/2/POWER = OFF
18:15:45 MQTT: stat/pete/2/POWER = ON

arendst, 2016-11-12:
Strange. My dual only does it once; Toggle 1 controls red, Toggle 2 controls green. I'll dive in the code tomorrow and see what it could be...

scargill, 2016-11-12:
ta

arendst, 2016-11-12:
Pete, what browser did you use when the dual failed?

scargill, 2016-11-12:
I would not say failed - more like "has a strange characteristic".  I am on Windows 10 - an up to date Chrome Browser.

Ok - so I tried a phone - and this does not happen. I'm stretching my memory but are you looking for stuff in the header because I seem to recall that with some browsers you see stuff twice - that would account for this happening twice on the PC (Windows 10 Chrome) and once on the phone (Android Chrome).  It's quite repeatable.

scargill, 2016-11-12:
Erm did I miss something - is there a way to put a password on the web page... seems to me that there's a minor risk here when picking a random address in the house could pull up the menu...? If not - any chance you could add one in - or turn it off... maybe 8 short presses? Or did I miss something again?

arendst, 2016-11-12:
Just execute command webserver 1 and all you see is just the button. If still feeling anxtious execute command webserver 0 and it's off.

If paranoia just disable define USE_WEBSERVER in user_config, recompile and upload from your ota webserver using upgrade 1.

scargill, 2016-11-12:
Ok - I can fix that - erm DUAL - I think I am right in saying you have relay 1 and 2 the wrong way around. If I press the hardware button twice - what Sonoff call relay1 goes on - if I press the button once, the relay Sonoff call relay 2 comes on. checking commands now (I've wired it into the new (400w) heater in the house - and I'm working in the office - not one of my better ideas).

scargill, 2016-11-12:
Quick one - POWER turns on and off the first relay... erm -what turns the second one on and off...  clearly missing something but I'm not seeing it in the docs.

scargill, 2016-11-12:
Actually - POWER turns the SECOND one on.

Also - I went through to the house - pressed the button 4 times expecting to see the display I saw originally - and all I got was setup for the web.  So... if you're at the unit and you've forgotten the settings and need to test something - how do you do that? Is there a way to get up that full menu?

arendst, 2016-11-12:
Well the whole idea of disabling the webpage is security right? So enabling it needs to bring some kind of burden. Send MQTT command cmnd/pete/webserver 2 and you'll be in business.

Btw you also access pete using the unique fallback topic name you wrote down when you installed the software right?

Relay 2 is simply accessed by cmnd/pete/2/power toggle as could relay 1 with command cmnd/pete/1/power toggle. This answer must have been obvious from your ON/OFF/ON problem.

scargill, 2016-11-12:
Yes, figured it all out - it's running the new heater on my node-red bigtimer.  Confirming both the button and the command are the wrong way around  /1/ turns on output 2 on the Sonoff dual.

Very good.

arendst, 2016-11-13:
Fixed no serial/web response when MQTT fails to connect in version 2.0.12 just released

ryan0930, 2016-11-13:
I am trying to upload sketch via the Arduino IDE for the Sonoff dual.  But always get the following errors.

Have connected the pins as below:
VCC ---> FTDI 3.3V
Rx ----> Tx FTDI
Tx ----> Rx FTDI
GND Vertical & GND (Button1 and Button2) ----> GND

Connection:  Press button and insert FTDI.

Are the above connection correct?

If yes, while uploading sketch, i get the following errors.  

*****
Arduino: 1.6.12 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Serial, All"

Build options changed, rebuilding all

Sketch uses 337,140 bytes (35%) of program storage space. Maximum is 958,448 bytes.
Global variables use 41,636 bytes (50%) of dynamic memory, leaving 40,284 bytes for local variables. Maximum is 81,920 bytes.
C:\Users\Robotics\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9/esptool.exe -vv -cd ck -cb 115200 -cp COM11 -ca 0x00000 -cf C:\Users\Robotics\AppData\Local\Temp\arduino_build_777930/sketch_nov13a.ino.bin 
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM11
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
warning: espcomm_sync failederror: espcomm_open failederror: espcomm_upload_mem failederror: espcomm_upload_mem failedThis report would have more information with"Show verbose output during compilation"option enabled in File -> Preferences.
****

Can someone help as i am a bit new to Sonoff.

Thank you

arendst, 2016-11-13:
Connections are fine but button won't work for Dual. You'll need a wire to short a Via a shown in the picture on the wiki Compile and Upload.

ryan0930, 2016-11-13:
First of all big thanks for the reply. Is it the same procedure as available in the below link (PIN 15 or EN_FW connecting to GND of horizantal pin connectors)?

http://support.iteadstudio.com/support/discussions/topics/11000006870

Please confirm.

Thank you.

[6 more replies]

closed 2016-11-16 by arendst