arendst/Sonoff-MQTT-OTA-Arduino #81: Enhancement - Save SD

scargill, 2016-11-13:
It just occurred to me - normally when I save parameters - such as the state of an output - to SD - I only do it if the value has changed.  But I use the SDK... Are you doing this - the reason I ask - I update values every minute....  if you actually forcibly write the state of the outputs to SD every time this may affect SD life. I BELIEVE there is an update command in Arduino but never used it - just checking in case.

Also with your new update comes first use of OTA and I've realised that like others not using Arduino - I've no idea where the BIN file(s) would be to put on my server for OTA - got a short guide anywhere? Do you keep bin files - or must we recompile and load up files to our own servers?

arendst, 2016-11-13:
I calculate a hash over the parameters and when a parameter changes they will be saved to one of two alternating flash sectors. This way wear is 50% less. I don't see why you would change any of my parameters every minute other than the state of the relay. When flash wear becomes an issue doing it this way I could always change the software to select another region of flash. There is plenty!

Regarding OTA I made some changes to two standard IDE files and add a python file allowing to select the location of the compiled file to be either serial (normal IDE behaviour) or to your OTA web server. See the arduino folder on github and the documentation on wiki (prerequisite).

I do not keep the binaries myself as they are different for Sonoff, Pow and ElectroDragon.

The standard location would be http://<your server>/api/arduino/<binary file> as can be seen in user_config.h (define OTA_URL).

scargill, 2016-11-13:
It is just the state of the relay that I change. Yep, spotted where the files are stored thanks. I'm asssuming I send the one bin file to my web server (which already serves bin files) - and point to that.

arendst, 2016-11-13:
Correct

closed 2016-11-16 by arendst