arendst/Sonoff-MQTT-OTA-Arduino #82: Serial since last update

scargill, 2016-11-13:
I don't know if it is the last update (grabbed this morning)- or because I'm now programming one of the older style sonoffs - but after installing the software, it is fine if I use the web interface.... but if I type serial - it accepts the first input - then dies - won't respond to serial, won't respond to web interface.

arendst, 2016-11-13:
I suppose you compiled for sonoff and NOT pow.

The default user_config.h probably does not contain your wifi and mqtt configuration so what happens is:

- the device will try to connect to wifi for 16 seconds which fails
- it starts WPSconfig allowing you to configure wifi using your routers WPS functionality for another 16 seconds
- if not connecetd reboots and starts again

If you want WifiManager instead of WPSconfig you would execute command smartconfig 2; device reboots and after 16 seconds start a wifi AP with IP address 192.168.4.1 for 1 minute. Then reboots again.

I still don't get the "ok" message as it's not in my software...

What serial command did you use?

kiralikbeyin, 2016-11-14:
@arendst I dont have a wps button my router. Is there any way : reboot for WifiManager after these fails?

arendst, 2016-11-14:
Have a look at the marvelous wiki (RTFW). Select Button and read it.

You might find that pressing the button quickly three times will start Wifi smartconfig and pressing it four times it even starts Wifi manager...

What a wonderful product ;-)

kiralikbeyin, 2016-11-14:
@arendst i read it 4 times button ok. 
How can i add one more chain to this process;

the device will try to connect to wifi for 16 seconds which fails
it starts WPSconfig allowing you to configure wifi using your routers WPS functionality for another 16 seconds
if not connected reboots and starts **WITH WIFIMANAGER :)**

arendst, 2016-11-14:
I just fooled around with the button and noticed that it doesn't do what I expected... I'll look into it.

As an alternative just enter command ```wificonfig 2``` to start wifimanager or ```wificonfig 1``` to start smartconfig.

As another alternative just enter command ```ssid <your ssid>``` and ```password <your wifi password>``` and your good to go.

kiralikbeyin, 2016-11-14:
@arendst ok thanks.

I have another problem, i cant upload with arduino ide or platformio. I am trying it 15-20 times everytime until it uploads. USB port is lost after 2-3 plug-unplug the ftdi (so i must restart mac).

I always press the button and plug the usb as described @scargill 's site :

>  Use FTDI at 3v3 setting – in the original boards,  press the large button –THEN power up FTDI.

arendst, 2016-11-14:
Unplugging and plugging USB probably won't work without restarting the IDE at least that's what happens in my case on Windows.

Why don't you keep the FTDI via USB connected to your Mac and:
- disconnect the 3V3 power lead from the FTDI cable to sonoff
- hold the button down
- connect the 3V3 power lead from the FTDI cable to sonoff

Mind you, this button thing will not work for a Sonoff Dual as the button is not connected to GPIO0. In that case see my wiki again (Compile and Upload)

scargill, 2016-11-14:
Hi there - sorry for the delay - I've been away..  YES I did compile for Sonoff and not POW but thanks for that timely reminder as I'd blown a POW without recompiling!!! No - this is a bog-standard old-shape Sonoff (not TH10 or 16) - any serial - as soon as you hit ENTER - it seems to go away for a while and eventually come back - more serial does the same - and yes I'd altered the credentials to suit my system.  Later today I'll try again and also test on a TH16... but this was not happening before the last update.

scargill, 2016-11-15:
So today I went back to using the TH16. I took a FRESH copy of your software updated 2 days ago - put my details in (MQTT,WIFI) and flashed the board. I then issued a reset 2 via serial.

It accepted the credentials and signed in.

HOWEVER - using "power off" - that works serially - put it in again - nothing - no output from the serial.
11 seconds later.... it acknowledges the power off  and will accept power off...
14 second later... etc.

11:04:41 power off
11:04:54 MQTT: stat/sonoff/POWER = OFF
11:05:15 power off
11:05:29 MQTT: stat/sonoff/POWER = OFF

This is all since the last update - I could fire correct and incorrect serial at it prior to that.

Can you fix this - also a suggestion - if you get an invalid command via serial, some kind of error message "Eh?" would be great..

Can't really do much now until this is sorted.

arendst, 2016-11-15:
Pete,

strange... I suggest to execute command ```seriallog 2``` first. This enables serial output which is initially disabled except for some time info.

For debugging purposes ```seriallog 3``` even outputs more information and may allow to show you why you receive such delays.

scargill, 2016-11-15:
I just tried this (and I've now tried this with the latest software on two devices - an orginal Sonoff and a TH16. I've even shorted the serial on my FTDI to make sure I wasn't going mad or my PC bust - instant response..

But on the boards.

00:00:10 HTTP: Webserver active on sonoff-6821 with IP address 192.168.0.120
00:00:10 MQTT: Attempting connection...
00:00:10 MQTT: Connected
00:00:10 MQTT: stat/sonoff/INFO = Sonoff module version 2.0.13, Fallback DVES_0CDAA5, GroupTopic sonoffs
00:00:10 MQTT: stat/sonoff/INFO = Webserver active for Admin on sonoff-6821 with IP address 192.168.0.120
00:00:10 MQTT: stat/sonoff/WARNING = Change MQTT_MAX_PACKET_SIZE in libraries/PubSubClient.h to at least 360
11:22:35 MQTT: stat/sonoff/1/POWER = OFF
11:22:44 power off
11:22:58 RSLT: stat/sonoff/POWER = OFF
11:22:58 power off

Note the Power off command at 11.22.44 and nothing back until 11.22.58 during which time no serial commands would be accepted.

MQTT commands work at full speed.. see times below.

11:28:44 MQTT: stat/sonoff/POWER = ON
11:28:47 MQTT: stat/sonoff/POWER = OFF
11:28:50 MQTT: stat/sonoff/POWER = ON
11:28:54 MQTT: stat/sonoff/POWER = OFF
11:29:04 grouptopic
11:29:18 RSLT: stat/sonoff/GROUPTOPIC = sonoffs
11:29:18 MQTT: Attempting connection...
11:29:18 MQTT: Connected
11:29:35 seriallog 3
11:29:49 MQTT: stat/sonoff/SERIALLOG = 3
11:29:50 Config: Saved configuration (692 bytes) to flash at FA and count 13
11:29:58 power off
11:30:11 MQTT: Receive topic cmnd/sonoff/power, data off
11:30:11 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
11:30:11 RSLT: stat/sonoff/POWER = OFF
11:30:12 MQTT: Attempting connection...
11:30:12 MQTT: Connected
11:30:49 power off
11:31:03 MQTT: Receive topic cmnd/sonoff/power, data off
11:31:03 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
11:31:03 MQTT: stat/sonoff/POWER = OFF

Not seeing anything in serial logging that gives the game away.

arendst, 2016-11-15:
Most worrying are the last four lines...

I cannot reproduce. I've downloaded from github too, compiled and uploaded without problem. 

I see you seem to have problems with MQTT (Messages with RSLT instead of MQTT caused by either no MQTT connection and/or the fact that you have MQTT_MAX_PACKET_SIZE not set to 400 as shown at 00:00:10 which results in messages larger than 128 not being send by pubsubclient).

Could you please try to execute command ```status 0``` and return the information?

arendst, 2016-11-15:
What I do see is that when there is no MQTT connection the software tries to reconnect every 10 seconds. During the first seconds of that attempt no serial input is acknowledged:
```
12:13:14 MQTT: Attempting connection...
12:13:20 MQTT: CONNECT FAILED, rc -2. Retry in 10 seconds
12:13:20 power off
12:13:20 MQTT: Receive topic cmnd/sonoff/power, data off
12:13:20 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
12:13:20 RSLT: stat/sonoff/POWER = OFF
12:13:21 Config: Saved configuration (692 bytes) to flash at FA and count 11
```
At 12:13:15 I executed command ```power off``` but it was acted upon after the MQTT connect fail message (obvious as pubsubclient tries to reconnect which seems to take 6 seconds and all else is halted) at 12:13:20. But then still the result came immediately.

It's OK with me but it's not the problem you experience...

scargill, 2016-11-15:
So here - I've rebooted the board - there is no MQTT data going back and forth - only serial and only me sending "power off" by serial every now and then.

19:02:55 MQTT: Receive topic cmnd/sonoff/power, data on
19:02:55 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data on (ON)
19:02:55 MQTT: stat/sonoff/POWER = ON
19:02:55 Config: Saved configuration (692 bytes) to flash at F9 and count 16
19:02:58 power off
19:03:05 MQTT: Receive topic cmnd/sonoff/power, data off
19:03:05 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
19:03:05 MQTT: stat/sonoff/POWER = OFF
19:03:05 power off
19:03:12 MQTT: Receive topic cmnd/sonoff/power, data off
19:03:12 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
19:03:12 MQTT: stat/sonoff/POWER = OFF
19:03:12 power off
19:03:19 MQTT: Receive topic cmnd/sonoff/power, data off
19:03:19 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
19:03:19 MQTT: stat/sonoff/POWER = OFF
19:03:19 Config: Saved configuration (692 bytes) to flash at FA and count 17
19:03:34 power off
19:03:41 MQTT: Receive topic cmnd/sonoff/power, data off
19:03:41 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
19:03:41 MQTT: stat/sonoff/POWER = OFF
19:03:41 power off
19:03:48 MQTT: Receive topic cmnd/sonoff/power, data off
19:03:48 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
19:03:48 MQTT: stat/sonoff/POWER = OFF
19:04:08 power off
19:04:14 MQTT: Receive topic cmnd/sonoff/power, data off
19:04:14 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
19:04:14 MQTT: stat/sonoff/POWER = OFF
19:04:14 power off
19:04:21 MQTT: Receive topic cmnd/sonoff/power, data off
19:04:21 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
19:04:21 MQTT: stat/sonoff/POWER = OFF
19:04:21 power off
19:04:28 MQTT: Receive topic cmnd/sonoff/power, data off
19:04:28 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
19:04:28 MQTT: stat/sonoff/POWER = OFF
19:04:28 power off
19:04:35 MQTT: Receive topic cmnd/sonoff/power, data off
19:04:35 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
19:04:35 MQTT: stat/sonoff/POWER = OFF
19:04:35 power off
19:04:42 MQTT: Receive topic cmnd/sonoff/power, data off
19:04:42 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
19:04:42 MQTT: stat/sonoff/POWER = OFF

So same issue - several seconds delay - but there is nothing wrong with the MQTT because if I DO send from Mosquitto...  I can turn that light on and off at any speed I like..

scargill, 2016-11-15:
Here is status 0  - again note the delay (I wiped my password from this but otherwise it's a straight grab.

19:07:38 status 0
19:07:45 MQTT: Receive topic cmnd/sonoff/status, data 0
19:07:45 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type STATUS, Data 0 (0)
19:07:45 MQTT: stat/sonoff/STATUS = 2.0.13, 1, sonoff, 0, POWER, 0, 1, 1, 1, 1, 0
19:07:45 RSLT: stat/sonoff/STATUS = PRM: Baudrate 115200, GroupTopic sonoffs, OtaUrl http://domus1:80/api/arduino/sonoff.ino.bin, Uptime 0 Hr, BootCount 7, SaveCount 17
19:07:45 MQTT: stat/sonoff/STATUS = FWR: Version 2.0.13, Boot 4, SDK 1.5.3(aec24ac9)
19:07:45 MQTT: stat/sonoff/STATUS = LOG: Seriallog 3, Weblog 2, Syslog 0, LogHost domus1, SSId beyond, Password 1921682012, TelePeriod 300
19:07:45 MQTT: stat/sonoff/STATUS = MEM: Program size 377kB, Free 560kB (Heap 30kB), Spiffs start 940kB (64kB), Flash size 1024kB (1024kB)
19:07:45 RSLT: stat/sonoff/STATUS = NET: Host sonoff-6821, IP 192.168.0.34, Gateway 192.168.0.1, Subnetmask 255.255.255.0, Mac 5C:CF:7F:0C:DA:A5, Webserver 2, WifiConfig 3
19:07:45 RSLT: stat/sonoff/STATUS = MQT: Host 192.168.0.20, Port 1883, Client DVES_%06X (DVES_0CDAA5), User admin, Password ********, MAX_PACKET_SIZE 128, KEEPALIVE 15
19:07:45 RSLT: stat/sonoff/STATUS = TIM: UTC Tue Nov 15 18:07:45 2016, Local Tue Nov 15 19:07:45 2016, Start DST Sun Mar 27 02:00:00 2016, End DST Sun Oct 30 03:00:00 2016

arendst, 2016-11-15:
Woh, 6/7 seconds delay. Nice problem. 
Status looks fine.
I'll investigate tomorrow.

scargill, 2016-11-15:
Thank you!!! Really appreciated –would save me some DEV work to use your code for POW units. Why re-invent the wheel.

arendst, 2016-11-16:
Pete, I cannot reproduce your serial problem so I need your help.

I made a version with debug information avialable here [https://github.com/arendst/Experimental/tree/master/SerialDebug](https://github.com/arendst/Experimental/tree/master/SerialDebug).

To allow me to gather as much info as possible please compile this version with your changes to user_config.h and upload. Then execute command ```seriallog 3``` and execute commands like you did before.

My response is like this:
```
11:58:29 power off
11:58:29 DBG: ++ Wifi test addLog
11:58:29 DBG: ++ Leaving addLog
11:58:29 DBG: ** Completed first addLog
11:58:29 DBG: ** Entered do_cmnd with "power off"
11:58:29 DBG: ** Starting mqttDataCb
11:58:29 DBG: ** Entered mqttDataCb
11:58:29 MQTT: Receive topic cmnd/sonoff/power, data off
11:58:29 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
11:58:29 MQTT: stat/sonoff/POWER = OFF
11:58:29 DBG: ** Completed do_cmnd
11:58:30 Config: Saved configuration (692 bytes) to flash at F9 and count 30
```
In your case I expect timing differences in one or more steps. This hopefully allows me to investigate further.

Thanks for your cooperation.

scargill, 2016-11-16:
Sorry about the delay - just seen this now - didn't get the email copy - right - working on it

scargill, 2016-11-16:
Ok here it is from power up all well until I hit a serial command "power off"

00:00:01 Wifi: STATION_IDLE
00:00:03 Wifi: STATION_IDLE
00:00:04 Wifi: STATION_IDLE
00:00:05 Wifi: STATION_IDLE
00:00:06 HTTP: Webserver active on sonoff-6821 with IP address 192.168.0.21
00:00:06 MQTT: Attempting connection...
00:00:06 MQTT: Connected
00:00:06 MQTT: stat/sonoff/INFO = Sonoff module version 2.0.13a, Fallback DVES_0CDAA5, GroupTopic sonoffs
00:00:06 MQTT: stat/sonoff/INFO = Webserver active for Admin on sonoff-6821 with IP address 192.168.0.21
00:00:06 MQTT: stat/sonoff/WARNING = Change MQTT_MAX_PACKET_SIZE in libraries/PubSubClient.h to at least 360
00:00:07 MQTT: stat/sonoff/1/POWER = OFF
00:00:08 RTC: (UTC) Wed Nov 16 13:18:25 2016
00:00:08 RTC: (DST) Sun Mar 27 02:00:00 2016
00:00:08 RTC: (STD) Sun Oct 30 03:00:00 2016
14:19:07 power off
14:19:07 DBG: ++ Wifi test addLog
14:19:21 DBG: ++ Leaving addLog
14:19:21 DBG: ** Completed first addLog
14:19:21 DBG: ** Entered do_cmnd with "power off"
14:19:21 DBG: ** Starting mqttDataCb
14:19:21 DBG: ** Entered mqttDataCb
14:19:21 MQTT: Receive topic cmnd/sonoff/power, data off
14:19:21 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
14:19:21 RSLT: stat/sonoff/POWER = OFF
14:19:21 DBG: ** Completed do_cmnd
14:19:21 power off
14:19:21 DBG: ++ Wifi test addLog
14:19:35 DBG: ++ Leaving addLog
14:19:35 DBG: ** Completed first addLog
14:19:35 DBG: ** Entered do_cmnd with "power off"
14:19:35 DBG: ** Starting mqttDataCb
14:19:35 DBG: ** Entered mqttDataCb
14:19:35 MQTT: Receive topic cmnd/sonoff/power, data off
14:19:35 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
14:19:35 RSLT: stat/sonoff/POWER = OFF
14:19:35 DBG: ** Completed do_cmnd
14:19:36 MQTT: Attempting connection...
14:19:36 MQTT: Connected

and now with your "seriallog 3"

from scratch

00:00:03 Wifi: STATION_IDLE
00:00:05 Wifi: STATION_IDLE
00:00:06 Wifi: STATION_IDLE
00:00:07 Wifi: STATION_IDLE
00:00:08 Wifi: STATION_IDLE
00:00:09 Wifi: STATION_CONNECT_FAIL
00:00:09 WPSconfig: Active for 1 minute
00:00:24 WPSconfig: FAILED with status 4
00:00:28 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
?
00:00:00 Config: Loaded configuration from flash at F9 and count 22
00:00:00 APP: Bootcount 13
00:00:00 Wifi: Patch issue 2186
00:00:00 Wifi: Connecting to beyond (1921682012) in mode 11N as sonoff-6821
00:00:00 APP: Project sonoff (Topic sonoff, Fallback DVES_0CDAA5, GroupTopic sonoffs) Version 2.0.13a
00:00:00 Config: Saved configuration (692 bytes) to flash at FA and count 23
[... 94 lines cut ...]
14:24:03 DBG: ++ Wifi test addLog
14:24:17 DBG: ++ Leaving addLog
14:24:17 DBG: ** Completed first addLog
14:24:17 DBG: ** Entered do_cmnd with "power off"
14:24:17 DBG: ** Starting mqttDataCb
14:24:17 DBG: ** Entered mqttDataCb
14:24:17 MQTT: Receive topic cmnd/sonoff/power, data off
14:24:17 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
14:24:17 RSLT: stat/sonoff/POWER = OFF
14:24:17 DBG: ** Completed do_cmnd
14:24:17 power off
14:24:17 DBG: ++ Wifi test addLog
14:24:32 DBG: ++ Leaving addLog
14:24:32 DBG: ** Completed first addLog
14:24:32 DBG: ** Entered do_cmnd with "power off"
14:24:32 DBG: ** Starting mqttDataCb
14:24:32 DBG: ** Entered mqttDataCb
14:24:32 MQTT: Receive topic cmnd/sonoff/power, data off
14:24:32 MQTT: DataCb Topic sonoff, Group 0, Device 1, Type POWER, Data off (OFF)
14:24:32 RSLT: stat/sonoff/POWER = OFF
14:24:32 DBG: ** Completed do_cmnd
14:24:32 MQTT: Attempting connection...
14:24:32 MQTT: Connected
14:24:52 power off
14:24:52 DBG: ++ Wifi test addLog

arendst, 2016-11-16:
OK, I can see it hangs while checking the wifi connection in support.ino at line 678. 
```
  if ((WiFi.status() == WL_CONNECTED) && (loglevel <= sysCfg.syslog_level)) syslog(line);
```
Apparently Wifi.status() takes a lot of time the first time it is accessed (14:24:17) but as that line is also executed at 14:24:32 I'm a bit puzzled.

Pete. could you execute command ```syslog 1``` and try again to ```power on``` please.

arendst, 2016-11-16:
Eureka! I can reproduce! It has something to do with syslog. If I change the loghost to a non-existent host it does the same as your system.

Thanks. I now can test myself and will investigate

scargill, 2016-11-16:
Excellent - meanwhile I've been using the time to test my own code - and was just about to write in to say I can guarantee it isn't anything to do with the hardware or the location or the test rig etc as mine is working - but it looks like you're onto it.... well done - I will wait with "baited breath" to see the next release.  For some reason this thing isn't sending me emails..... I'll look in later.

scargill, 2016-11-16:
erm, when next you reply.. what's a loghost?

arendst, 2016-11-16:
A loghost is a linux syslog host allowing devices to send messages over UDP which can be logged in linux system syslog or a seperate log depending how you configure logging on your linux loghost.

I use it to monitor all (MQTT) messages from all ESP8266 based sonoffs/sensors/displays on my domus1 MQTT/Domoticz virtual server.

patbeirne, 2016-11-16:
If you wish, you can ask your Sonoff to send diagnostics to the "loghost" by sending ASCII strings through UDP to "loghost" on port SYS_LOG_PORT (defaults to 514). You can set the verbosity of these diagnostics with the command "syslog"

arendst, 2016-11-16:
Thnx guys for your support.

I've released version 2.0.15 which solves the serial problem introduced in version 1.0.17 (!) caused by an unavailable sysloghost.
With the knowledge of today I also fixed more possible syslog related problems.

I hope that finalises this issue...

scargill, 2016-11-16:
Yup, that seems to be that! Well done.

closed 2016-11-16 by arendst