arendst/Sonoff-MQTT-OTA-Arduino #84: Upgraded to 2.0.13 using OTA now Web UI is broken

tam481, 2016-11-14:
Hi,
I've managed to upgrade from 2.0.7 to the latest version 2.0.13 and now the web UI won't load! I can still use MQTT to switch the relay on and off but that's it now.

#define WEB_SERVER             2

I did not change the parameter value.

Is there anything I can do without having to disconnect and dismantle the Sonoff? It's connected to a wall socket in a hard-to-reach place behind a sideboard and getting it out isn't going to be easy not to mention the grief I will get from my better half about making a mess ;-)

arendst, 2016-11-14:
What you can do:

- Try another browser
- Execute MQTT command ```cmnd/sonoff/status 5``` to check IP address and status of webserver
- Stop and restart webserver with commands ```cmnd/sonoff/webserver 0``` followed by ```cmnd/sonoff/webserver 2```
- Restart sonoff with command ```cmnd/sonoff/restart 1```
- Execute command ```cmnd/sonoff/status 4``` and look at memory allocation (SPIFFS size and Flash size) to verify you compiled the new version with the correct IDE settings.

tam481, 2016-11-14:
Thanks @arendst 
How do I get the output from MQTT? I've tried running two putty sessions to the RPi, one subscribing to cmnd/# and the other publishing to cmnd/sonoff/status -m 5 and other commands you recommended above but all I get is the actual message I published. I used to be able to run commands via the web interface in the past. What am I missing?

patbeirne, 2016-11-14:
Tam, 

See this helper page https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/wiki/Sonoff-&-MQTT-for-Newbies

to get info from the Sonoff, you subscribe to "stat/sonoff/+" 

That last character is a wildcard. Some systems use "+" and some use "#"

arendst, 2016-11-14:
You'll missing the response from sonoff; Commands are send using cmnd/ and the response is send as ```stat/```. Try this if using mosquitto:

```mosquitto_sub -t stat/# -t tele/# -t cmnd/# -v```

Or get mqtt-spy, great MQTT tool everyone using MQTT should use.

tam481, 2016-11-14:
thank you @arendst & @patbeirne 

I've done a status 4 and got this
stat/sonoff/STATUS MEM: Program size 337kB, Free 600kB (Heap 38kB), Spiffs start 940kB (64kB), Flash size 1024kB (1024kB)

Is that correct? Also, the web site is "refusing" connections! 
I've tried to reset, reboot etc to no avail. Any suggestions? I'm running out of ideas.

arendst, 2016-11-14:
What's the result of status 5?

tam481, 2016-11-14:
stat/sonoff/STATUS NET: Host sonoff-5997, IP 192.168.1.43, Gateway 192.168.1.1, Subnetmask 255.255.255.0, Mac 18:FE:34:BB:E7:4D, Webserver 2, WifiConfig 3

arendst, 2016-11-14:
looks fine too. If you're still unable to open a webpage at 192.168.1.43 please dump all statusses with command status 0 and show it here.

Did the version change to 2.0.13?
Did you try an OTA upgrade?
Did you power cycle sonoff already?

tam481, 2016-11-14:
Many thanks to you both for your time and apologies for wasting it in a way. I found out what the problem was. I had accidentally commented out the "#define USE_WEBSERVER" in the user_config.h

I've managed to resolve the issue by recompiling and carrying out an OTA upgrade via MQTT and the OtaUrl which I hadn't done before and found extremely useful especially since I'm considering disabling the web interface entirely.

Many thanks for all your help and the great effort you've put into this project particularly your willingness to accommodate other people's requirements.

closed 2016-11-15 by arendst