arendst/Sonoff-MQTT-OTA-Arduino #85: Flashing problems Generic8266-Nodemcu

kiralikbeyin, 2016-11-14:
I am using mac sierra.
1.6.8 IDE with 1.6.10 avr boards lib (somehow 1.6.8 boards lib doesnt work for me)
2.3.0 stable
sonoff SV

When i try to upload sonoff ino with generic esp8266, it never ends to %100 dots are freezing when very near to %100.
But i can upload the sketch with nodemcu 4mb-3mb SPIFFS 

I have a general problem: When i plug-unplug ftdi 2-3 times, usb port is dissapearing (not related to sonoff). Is there anybody with this problem?

TLaren, 2016-11-14:
Pretty sure that they only have 1M flash. Set it to 1M/64K
Oh look, I can say it too, "Read the Wiki".
https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/wiki/Compile-and-Upload

kiralikbeyin, 2016-11-14:
But i can upload the sketch with nodemcu 4mb-3mb SPIFFS

I think there is a problem with current.
My temporary solution is writing with **faster** baud rate. Now i am using firmware upgrade from admin menu - a bin file which compiled with 115200 baud.

Now i have another problem;
I added my 433mhz receiver and when i press my remote controller on-off buttons i'm running;
      digitalWrite(REL_PIN, HIGH);
      digitalWrite(REL_PIN, LOW);

But mqtt and webserver doesnt change its on-off status.

How can i make correct status?

oriolism, 2016-11-21:
Did you manage to fix this problem? Im having the same issue with the settings recommended in the wiki.

kiralikbeyin, 2016-11-21:
Double check your ftdi to esp wires (use short wires if possible)
Plug-unplug VCC when you are pressing esp button. Try plug-unplug directly usb also.
Try restarting your pc.
Try plug-unplug all ftdi and esp wires again.
Try another ftdi card.
RC2.3.0 and 1.6.8

This lib gave me only an idea and info about gpio pins of son off. I am using my own code now because of heavy code which are all in front of you.

ekgerke, 2016-11-30:
Not starting a new topic, since the problem is flashing related as well. Got my new patch of sonoffs (new packaging, different flash memory chip), flashed them like usually. Plugged the board in to the ftdi (tx, rx, gnd, 3.3v) and booted them up to check functionality, then put them to flash mode and started to burn the firmware. 3 out of 7 boards got fried (the 8mbit flash chips had physical burn marks). Inspected the boards for problems around the modified part (newly soldered pin strips), couldn't find anything wrong there. Could it be that sonoff has implemented some kind of "flashing protection" to prevent users from modifying the boards? Googled around a little as well, but couldn't find anything useful.

arendst, 2016-11-30:
What kind of sonoff is it? Just Sonoff or Sonoff Pow? Did you connect FTDI 3V3 while also connected to AC?

iTead advertises with hacker friendly so I can't imagine they made it a destructive process.

ekgerke, 2016-12-03:
just a simplest sonoff. the mains were'nt connected at all, just the 5-pin header part (rx, dx, gnd, 3.3v). the process was the same as I have done multiple times before. 4 of the 7 modules flashed remained intact. besides the power and esp chip part of the board are still fine (serial monitor shows "garbage" when putting the microcontroller to boot mode). makes me wonder whether or not the connection to the mains would actually destroy the module as well. maybe the destroyed example was yet another defective board as were the three I had ordered.

closed 2016-12-27 by arendst