arendst/Sonoff-MQTT-OTA-Arduino #86: Additional domoticz-related functionality

ekgerke, 2016-11-14:
Hi.

I have modified sonoff hardware to accept an external spring loaded switch to gpio0 (i.e. standard EU size wall switch), so the sonoff itself could be hidden inside the wall and could be manually operated just like a regular light switch would. I also added a sensor plug in order to add a (dht22) sensor to monitor each room temperature/humidity. That way the sensor could be mounted away from the switch (i.e. outdoors, next to  vent grill, etc.).

While doing my own setup, I figured I would love to have a few additional features to the sonoff firmware. I could start my own, private fork of the code, of course, but I figured someone else could probably appreciate those changes aswell.

Additional domoticz functionality in domoticz conf page could include: 

1) idx for double click (relay2, button2 or whatever you might call it) - a virtual switch that could be used to trigger a scene. (for example double clicking gpio0 in the bedroom could trigger a good night scene which turns off all the lights in the house)

2) a configurable idx to publish dht22 to domoticz. I could easily add a sensor device to sonoff pin5 (gpio14) and map it to domoticz dummy device sensor module.

I realize you have made a python mqtt gateway for mqtt, but I think it would be easier to configure devices in one go.

I hope I made some sense.

Gerke.

arendst, 2016-11-15:
I'll look into it...

0n3-70uch, 2016-11-29:
Good idea. The second point is very important if you want to swtch and measure with one device at the same time.

Thanks.

ekgerke, 2016-12-16:
Hi.

I played around with the code a little and added the functionality for adding dht22 to domoticz. I tested adding the dht22 sensor to the sonoff that comes without the probe functionality and 5 out of 6 modules work just fine (the one not working is simply caught in restart loop. haven't connected the serial console, so I'm not sure what could cause it).

rlust, 2016-12-30:
How are you adding Temperature telemetry Temp & Humidity data to Domoticz? I have been trying with Node-Red with no luck.

ekgerke, 2017-01-07:
just a few notes on temp/humidity sensors.

domoticz takes humidity as an integer not as a float. stemp2 needs to be rounded in order to get a reasonable reading. also, the HUM_STAT for domotics is as follows:

(((stemp2 >= 0 && stemp2 <= 39)) ? 2 : (stemp2 >= 40 && stemp2 <= 70) ? 1 : (stemp2 > 70) ? 3 :  0);

where 2 is dry, 1 is comfortable, 3 is wet and 0 is normal.

the json format for domoticz is as follows:

PSTR("{\"idx\":%d, \"nvalue\":0, \"svalue\":\"%s;%d;%d\"}"), idx, temperature, humidity, humidity_status

hope this eases the development a little.

arendst, 2017-01-27:
NextGen Version 3.9.3 contains a lot of Domoticz updates like defining sensor indexes to supported sensors. See Domoticz Configuration webpage for more information.

closed 2017-01-27 by arendst

0n3-70uch, 2017-01-27:
Thanks a lot.