arendst/Sonoff-MQTT-OTA-Arduino #87: Question/suggestion regarding the pubsub prerequisite

battika, 2016-11-14:
Hello Theo,

Thank you for this well-documented and nicely written firmware for the Sonoff devices.

I have a question/suggestion regarding the pubsub prerequisite.

According to the wiki, the maximum packet size needs to be changed to 400 in the pubsubclient.h file like this
``` C
#define MQTT_MAX_PACKET_SIZE 400
```
However when I checked the original pubsubclient.h file I found the following snippet in it.
``` C
// MQTT_MAX_PACKET_SIZE : Maximum packet size
#ifndef MQTT_MAX_PACKET_SIZE
#define MQTT_MAX_PACKET_SIZE 128
#endif
```

So, in my opinion you could define MQTT_MAX_PACKET_SIZE in the sonoff.ino or the user_config.h file to override the one defined in the pubsubclient.h file. This way we would be able to use the original library in the Arduino IDE without manually downloading/modifying it. Just checked and it seems to work for me. 

Thank you for your consideration.

arendst, 2016-11-14:
Would be nice but...

If checked it and it fails to work. Only messages no longer than 128 bytes are being send; Larger messages are not received by the broker.

Are you able to execute command ```status 0``` and receive all status messages? I just receive the ones shorter than 128 bytes.

battika, 2016-11-14:
Hello Theo,

Without the #define I used to have a warning at startup saying I need to have it. It's now gone, so a good indication it's in effect.

After executing status 0 in the console I get the following
```
16:25:29 MQTT: stat/sonoffpow/STATUS = 2.0.13, 1, sonoffpow, 0, POWER, 0, 1, 1, 1, 1, 0
16:25:51 MQTT: stat/sonoffpow/STATUS = 2.0.13, 1, sonoffpow, 0, POWER, 0, 1, 1, 1, 1, 0
16:25:51 MQTT: stat/sonoffpow/STATUS = PWR: Voltage 0 V, Current 0.000 A, Current Power 0 W, Total Power Today 0.000 kWh, Power Factor 0.00
16:25:51 RSLT: stat/sonoffpow/STATUS = PWR: Threshold PowerLow 0 W, PowerHigh 0 W, VoltageLow 0 V, VoltageHigh 0 V, CurrentLow 0 mA, CurrentHigh 0 mA
16:25:51 RSLT: stat/sonoffpow/STATUS = PRM: Baudrate 115200, GroupTopic pows, OtaUrl http://domus1:80/api/arduino/sonoffpow.ino.bin, Uptime 2 Hr, BootCount 3, SaveCount 6
16:25:51 MQTT: stat/sonoffpow/STATUS = FWR: Version 2.0.13, Boot 31, SDK 1.5.3(aec24ac9)
16:25:51 RSLT: stat/sonoffpow/STATUS = LOG: Seriallog 2, Weblog 2, Syslog 2, LogHost 192.168.x,yyy SSId xyz, Password xyz, TelePeriod 300
16:25:51 RSLT: stat/sonoffpow/STATUS = MEM: Program size 380kB, Free 556kB (Heap 25kB), Spiffs start 940kB (64kB), Flash size 4096kB (1024kB)
16:25:51 RSLT: stat/sonoffpow/STATUS = NET: Host sonoffpow-6703, IP 192.168.x.yyy, Gateway 192.168.x.yyy, Subnetmask 255.255.255.0, Mac 5C:CF:7F:XX:YY:ZZ, Webserver 2, WifiConfig 2
16:25:51 RSLT: stat/sonoffpow/STATUS = MQT: Host 192.168.xxx.yyy, Port 1883, Client DVES_%06X (DVES_XXYYZZ), User xxxxxx, Password yyyyyyyy, MAX_PACKET_SIZE 400, KEEPALIVE 15
16:25:51 RSLT: stat/sonoffpow/STATUS = TIM: UTC Mon Nov 14 15:25:51 2016, Local Mon Nov 14 16:25:51 2016, Start DST Sun Mar 27 02:00:00 2016, End DST Sun Oct 30 03:00:00 2016
```

Where do you have your #define? I have it in the user_config.h file as it's included before the pubsubclient.

arendst, 2016-11-14:
As you defined it in sonoff.ino the warning is gone but pubsubclient still doesn't know you changed it...

You have the same error as me! Although it seems to work it just doesn't. As can be seen on the last line; it starts with RSLT instead of MQTT as can be seen on the sixth line from below. This means that the message is denied by pubsubclient and is NOT send to the broker.

battika, 2016-11-14:
Correct, you are right, I've just checked it using the mosquitto_sub command and those items are not sent... Let me see if I can come up with anything else...

battika, 2016-11-15:
Closing this thread as could not find anything :) Thanks for checking.

closed 2016-11-15 by battika