arendst/Sonoff-MQTT-OTA-Arduino #89: Possible bug, on/off URL

JoeTN, 2016-11-15:
Hello Theo

first of all I really appreciate your work on this :-)

I got your firmware uploaded and running, and everything is working nicely, but I think I've found something that could be a bug, even though I'm not sure.

I can use the following URL to toggle the relay both on and off: http://192.168.1.111/?o=1
- But shouldn't it be so that http://192.168.1.111/?o=1 will only toggle it on, and do nothing if it is already on, instead of toggling it off?

For me, it would make sense if http://192.168.1.111/?o=1 would toggle it on, and http://192.168.1.111/?o=0 would toggle it off. Currently sending the same URL is toggling it both on and off, as I said, I don't know if that's a bug or what.

Also I would like to suggest the following additions:
- Add ability to give the Sonoff relay a name and display it on the webpage for easily identification. This way I could easily name my Sonoff relay in the kitchen to be "Kitchen", and then see that name on the frontpage for easy identification.
- I was thinking it would be nice with some simple authentication form, prompting for login-credentials on the webpage. But I guess that would have influence on the trigger-URL as it would have to include username and password for authentication. I have no idea if that's too hard to implement.

Again thank you for your great effort! - If I can donate somewhere I would be happy to do so :-)

Regards
Joe

arendst, 2016-11-15:
Thnx Joe.

No, it's not a bug, it's a feature used to TOGGLE the relay addressed by the number.

To send a command using http you would use something like http://192.168/1/111/cm?cmnd=power%20on (%20 is a space character).

The name shown below "Sonoff module" is the wifi hostname you can be changed with the hostname command like ```hostname Kitchen```.

There will be no authentication on HTTP as the goal of this software is to use MQTT to control sonoff and NOT the webpage (=HTTP). The webpage is just a gadget to easily configure and test functionality. Once done disable webserver with ```webserver 1``` (user mode) or ```webserver 0``` (off) and control sonoff using home automation tools like node-red, domoticz, openhab, fhem etc.

JoeTN, 2016-11-15:
Aah, thanks for the clarification!
Nice feature with the hostname command, that worked great!

This is really great :-)

closed 2016-11-15 by arendst