arendst/Sonoff-MQTT-OTA-Arduino #90: HowTo: Domoticz question

Malarcy, 2016-11-15:
Hi - this looks great - very excited to play.

Noob questions:

I am very familiar with Domoticz - new to MQTT and to this project.

I can see how to setup MQTT domoticz running mosquitto and I can publish to the domoticz/in topic and control a virtual device (as per the domoticz wiki). 

The part I am missing is the domoticz/out controlling these (and other MQTT devices) - I see you have a number of projects - some of them have scripts for domoticz - but this particular project you say "native Domoticx support" - so - can you tell me what I need to do on the domoticz end (I currently have mosquitto, but not node-red installed) to create a device that controls the relays in the modified sonoff device(s). - or - point me to some online description of how to procede.

The domoticz wiki describes adding flows to node-red - your other projects seem to use a native (python?) mqtt library from a device script - but I feel like I am missing something. Any help appreciated.

Thanks in advance

arendst, 2016-11-15:
To use Domoticz you need a MQTT broker as I only support Domoticz via MQTT.

- So mosquitto is fine.
- Default Domoticz is enabled in user_config.h so no need to change anything.
- Configure a virtual switch in Domoticz and remember it's device idx
- Open sonoff webpage, select Configure Domoticz and enter the device idx in the appropriate field

That's it. Sonoff will reboot and send the current switch status to Domoticz on power up. 

No need for node-red or my domoticz scripts.

Malarcy, 2016-11-15:
thanks for the response - all makes sense. The one part I was missing was seeing the messages on the domoticz/out topic (I was "watching" all topics  with mosquitto_sub -t  /#) - I didn't see anything when toggling switches in domoticz - I switched to specifically mosquitto_sub -t domoticz/out and I can see the json payloads as domoticz goes about it's business.

Very cool.

Waiting for the sonoff's to arrive now so I can play properly.

Thanks again - and be sure to post if you decide on a donate facility - you deserve a pizza/beer.

closed 2016-11-15 by Malarcy