arendst/Sonoff-MQTT-OTA-Arduino #93: Configuring of new devices (TH10/TH16) with an external sensor (DHT/DS18b20)

f4n, 2016-11-16:
Hello everybody,
before i start... great project, very good project structure, it seems a professional was working on that 👍 

I'm currently thinking about configuring new sonoff devices, especially TH10/TH16. These devices have a headphone jack to attach a DHT or DS18b20 sensor. I like the idea of a single binary file which fits for all needs. This will make updates more easy. I found two solutions which i would like to discuss with you.

1) We extend the web frontend with a new configuring page to select the sensor type.
Pro: Switching the sensor will be easy
Con: Both libraries, the DHT and DS18b12, are always in the compiled binary (more space is used)

2) itead recommendes to change the sensors only on turned off devices. So a second solution is to do an auto-detection on power on. This should be possible by simply try to read a sensor. If it fails you try the the next one and so on. The first sensor which is successfully read will be kept for further reads.
Pro: Just power it on, nothing to do. Very Easy! ;)
Con: Same here, both libraries, the DHT and DS18b12, are always in the compiled binary (more space is used). Maybe the initializing and start of the device will be slower.

So what do you think? I will help to implement the best solution.

Regards, f4n

arendst, 2016-11-18:
A third and best solution is using the current version 2.0.16.

Make changes to user_config.h as needed, recompile and send the updated firmware to your sensor sonoff. 

No need to burden sonoff with additional webpages and dead code.

f4n, 2016-11-20:
Of course, this feature should be configurable (as all other features). Even if you just uncomment one of SEND_TELEMETRY_DS18B20 or SEND_TELEMETRY_DHT there is no (big) change in code. Just one library is compiled in the binary. But, and that is the feature, if you uncomment both you will be able to switch between these sensors. And you only compile it once and use it on any kind of node. More sensor types may follow and to compile a new binary with at least 95% the same code just to change the external sensor seems a lot of unnecessary work to me. Or what if a sonoff doesn't work properly and you want to replace it, you never have to think about uploading the proper binary.

arendst, 2016-11-20:
I suggest you take a look at ESPEasy [http://www.letscontrolit.com/](http://www.letscontrolit.com/) which has a host of sensors to choose from while I keep this software as simple as possible.

closed 2016-11-21 by f4n