arendst/Sonoff-MQTT-OTA-Arduino #94: messageformat

scargill, 2016-11-16:
Erm, I selected messageformat 1

Didn't seem to do much - the messages look the same

arendst, 2016-11-17:
It currently only supports status messages (ie ```status 4```).

arendst, 2016-11-18:
Now all telemetry data can be returned in JSON format.

TotallyInformation, 2016-11-19:
Can that be made a default by adjusting the user_config.h file?

It would fit in with my configuration much better.

arendst, 2016-11-20:
I will implement this but keep in mind that it will only work once as changes to user_config.h will by default not change any STORED values in flash.

By changing the value of  CFG_HOLDER flash will be reinitialized with default values from user_config.h

TotallyInformation, 2016-11-20:
Thanks. That's fine as I would just prefer to be able to set it as the default for new devices.

arendst, 2016-11-23:
Implemented in version 2.0.17 just released

arendst, 2016-12-08:
Version 3.0.0 supports full JSON messages.

closed 2016-12-17 by arendst