arendst/Sonoff-MQTT-OTA-Arduino #97: Bug? MQTT Topic Issue

TotallyInformation, 2016-11-19:
Hi, again many thanks for all your time and effort on this code, it is very much appreciated.

I needed to move all of the cmnd, stat and tele topics in order to fit in with my existing Node-RED based home automation project.

So I made some adjustments in user_config.h as follows:
```c++
#define SUB_PREFIX             "COMMAND/sonoff"       // Sonoff devices subscribe to:- cmnd/MQTT_TOPIC and cmnd/MQTT_GRPTOPIC
#define PUB_PREFIX             "DEVICES/sonoff"       // Sonoff devices publish to:- stat/MQTT_TOPIC
#define PUB_PREFIX2            "DEVICES/sonoff"       // Sonoff devices publish telemetry data to:- tele/MQTT_TOPIC/UPTIME, POWER/LIGHT and TIME
                                            // May be named the same as PUB_PREFIX
#define MQTT_TOPIC             MQTT_CLIENT_ID  //PROJECT
```

So I expected to get output on the topics:
```
DEVICES/sonoff/sonoff-5138/#
```
But this isn't the case. Instead I am getting output on:
```
DEVICES/sonoff/sonoff/#
```

Same problem with issuing commands `COMMAND/sonoff/sonoff/status` is accepted but `COMMAND/sonoff/sonoff-5138/status` is not. However `COMMAND/sonoff/DVES_073412/status` *is* accepted (which is defined by `DEF_MQTT_CLIENT_ID` in the main code.

*So am I right in thinking that there is a code issue somewhere or have I misunderstood the settings?* I have several Sonoff's that I want to set up so I need the MQTT topics to be unique & I would prefer that do come from the code so that I don't have to remember to reset them by hand after updates.

Regards, Julian.

TotallyInformation, 2016-11-19:
As an aside, I also note a small issue on the /mq page of the webserver. The Client ID is listed there as:
```
Client Id (sonoff_000356)
```
Even though the startup log message gives me:
```
00:00:07 HTTP: Webserver active on sonoff-5138 with IP address 192.168.10.10
```
I tried `COMMAND/sonoff/sonoff_000356/status 5` which also doesn't work. Also the default entry in the input box is
```
DVES_%06X
```
e.g. the setting in the main code and not the setting in `user_config.h`

Hmm, OK so further down I see:
```
Topic (sonoff_%06X)
sonoff
```
I'd expect the part in brackets to be the current entry (`sonoff`) to match the other entries and the part in the input box to match the `MQTT_TOPIC` entry in `user_config.h` as the comments in that file seem to suggest.

**OK, I think I've spotted part of the problem now**. The `MQTT_TOPIC` isn't interpreting the `%06X` part of the `MQTT_CLIENT_ID` that I've set. So I guess that it is picking up some kind of default further down the code.

Ah, another related issue too. The MQTT user needs to be unique otherwise you can get unexpected crashes. So it would be better if it were able to include a unique code related to the mac address or device id so that it would be automatically unique without user intervention.

So I guess the final question is: *What is the best way to automatically generate unique topics and MQTT user names**

arendst, 2016-11-20:
Currently SUB_PREFIX, PUB_PREFIX and PUB_PREFIX2 are only supported as ONE word WITHOUT a slash as that makes it two or more words for MQTT.

Doing it that way you should not encounter the problems you see.

Background: The software uses the amount of slashes to distinguish between Subscription (SUB_PREFIX), Device (TOPIC) and Command (Rest of MQTT topic). So only MQTT topics like SUB_PREFIX/TOPIC/Command are recognized.

arendst, 2016-11-20:
The MQTT_User and MQTT_Topic are equal for all sonoffs. The MQTTClient is unique and is using the last 6 characters of the MAC address.

You can change MQTT_User for a sonoff using the MQTTUser command although I have no idea why you would.
You can and should change the MQTT_Topic for a sonoff using the Topic command but you can always access the sonoff by the fallback topic being MQTTClient.

arendst, 2016-11-26:
Implemented multilevel SUB_PREFIX allowing constructs like "COMMAND/sonoff" in version 2.0.18 just released

TotallyInformation, 2016-11-26:
Genius! Thanks Theo, you're a hero.

closed 2016-11-26 by arendst