arendst/Sonoff-MQTT-OTA-Arduino #98: Not able to update firmware OTA

drhelmer, 2016-11-21:
I have happily started using this software, and it is really,really great! However I tried using the webserver to upgrade the firmware with the compiled (Platformio) .bin file, but I get an error :

File size is larger than available free space
Update error code (see Updater.cpp) 8

The file I try to upload is the .pioenvs/sonoff/firmware.bin 

The file size is only 374 Kb so shouldnt be any problem I guess, and is the same that is used uploading  via serial.

Any idea what goes wrong?

arendst, 2016-11-21:
How much free space do you have according to the webserver information page?

drhelmer, 2016-11-21:
This is what the webserver says : Flash size	1024kB
Program flash size	4096kB
Program size	374kB
Free program space	2696kB
Free memory	26kB

arendst, 2016-11-21:
So what do you see:
- The physical flash size is 1024k
- You compiled the current code for 4096k flash. It loaded but it's memory lay-out is wrong
-  Free memory is way too small

Solution: recompile for 1024k flash with 64k spiffs and upload via serial interface.

arendst, 2016-11-22:
I did some testing and it seems the Platform IO parameters were not valid for a 1M sonoff. I just uploaded the correct version so please use the updated ```platformio.ini``` file.

drhelmer, 2016-11-22:
I updated the platformio.ini file, and this time the upload seemed to work, as I clicked on start upload, some time passed, I got upload succeeded, and the device rebooted, and the webserver started again. BUT I had changed the OTAURL in the user_config.h to this #define OTA_URL              "http://192.168.0.4/" PROJECT ".ino.bin", but the webserver still showed the domus1:80 URL. To be certain I ran strings utility on  firmware.bin, that I tried to OTA upload, and it showed clearly that the fiormware contained a new OTA string : strings firmware.bin |grep sonoff.ino.bin
http://192.168.0.4/sonoff.ino.bin

So the OTA upload from File seems to work, but dont, I havent tried from URL yet.

drhelmer, 2016-11-22:
Sorry I wasnt aware that the variables in user_config.h was stored in flash, even after a recompile. After I ran a reset in the webserver GUI the correct URL was displayed for OTA!

Great. I love you work, a little gem :)

closed 2016-11-22 by drhelmer