arendst/Sonoff-MQTT-OTA-Arduino #99: Unable to Upload Sketch

oriolism, 2016-11-21:
I setup the Arduino IDE versions 1.6.12 with the setting recommended on the wiki but the upload doesn't complete.

Sketch uses 382,370 bytes (39%) of program storage space. Maximum is 958,448 bytes.
Global variables use 40,380 bytes (49%) of dynamic memory, leaving 41,540 bytes for local variables. Maximum is 81,920 bytes.
Uploading 386528 bytes from /var/folders/20/7mblxn_j2cs0nsjjc3_fmwwc0000gn/T/arduino_build_259410/sonoff.ino.bin to flash at 0x00000000
................................................................................ [ 21% ]
................................................................................ [ 42% ]
................................................................................ [ 63% ]
................................................................................ [ 84% ]
.........................................................

Any recomendation to fix this issue?

patbeirne, 2016-11-21:
Oriolism,
As strange as it sounds, try increasing the baud rate from the Arduino IDE to the ESP up to 230400 baud. This makes a huge difference for me; the downloads run to 100% almost every time.

oriolism, 2016-11-21:
@patbeirne Already tried all the baud rates without successful.

arendst, 2016-12-17:
See #85

closed 2016-12-17 by arendst