arendst/Sonoff-MQTT-OTA-Arduino #100: Security Bug

Beetle-II, 2016-11-23:
Hello.
I change webserver to mode 1 (user mode), but i can enter to page http://IP/cm

arendst, 2016-11-23:
Fixed in version 2.0.17 just released

mactsolutions, 2016-11-24:
I tried 2.0.17 but it still has the security bug discussed here.

arendst, 2016-11-24:
If I execute command ```webserver 1``` then all I can see is the user allowed screens, all others are redirected to this screen.

Please explain exactly the steps you take opening the console page (http://IP/cm) while in in User Mode.

mactsolutions, 2016-11-24:
its now in user mode when I executed it in the console. My expectation was that it will be in user mode once i flush the script with #define WEB_SERVER  1 in the config file.

arendst, 2016-11-24:
The settings in the config file are used initially and saved in flash. Changing the settings in user_config.h afterwards has no effect on the settings in flash UNLESS you change the define CFG_HOLDER.

Most settings can be changed using commands from the extensive command list. See Wiki.

closed 2016-11-24 by arendst