arendst/Sonoff-MQTT-OTA-Arduino #101: HC-SR501 Human Infrared PIR Motion Sensor Detector

littlebilly, 2016-11-24:


alphalove, 2016-11-25:
Hello littlebilly,
I've just implemented a changed to the code thatmight meet your needs.  My need was to wire in standard house wall switches so they can still control the lights (step at a time with the wife!).  The code also allows you to modify the behaviour of the input (follow the input on GPIO14 or toggle on each change) via an MQTT command (switchmode as i've called it).

I'll attempt to submit this as a pull request and see if arendst is willing to have it as part of the main code base.

I've also ordered a bunch of these (http://www.ebay.com.au/itm/161999859668) which I'll cut and use to interface to GPIO14.

Also on the TH10's that I've got, the other I/O is available on the 2.5mm socket, it's just itead haven't installed the 0 Ohm input and 10k pull-up resistors.

Cheers,

alex

littlebilly, 2016-11-25:
Hello alex,
thanks a lot for your answer. As far as I got it your code might meet my needs.
The general idea is to use the input on GPIO14 as a simple alarm input triggered by an on/off sensor
like this HC-SR501 or a simple reed-contact. This should trigger a MQTT command.

Cheers,
Billy

mihalski, 2016-11-25:
It would be a nice additional feature to have in this firmware. I know there is an alternative that does this now, but I have not personally tried it:

https://github.com/mertenats/Sonoff

arendst, 2016-11-25:
Now that is usefull! I'll look into it.

tam481, 2016-11-25:
@alphalove. Could you please elaborate on how you managed to wire in the standard wall switch. I'm having the same issue with my better half who won't let me do anything to the wall switches.

alphalove, 2016-11-25:
Hello tam481,
Essentially the TH10/16 replaces each light/switch junction box in the roof space.  In the junction box the permanent AC supply is wired to the input of the TH10/16 (i.e the TH10/16 is always powered) the light is wired to the relay output, and the wall switch is wired to GPIO14.  The TH10/16 already has a pull-up resistor on GPIO14, so one side of the wall switch is connect to GPIO14 and the other to the TH10/16 ground.  As mention above I've bought some 2.5mm leads from ebay which I'll cut in half and wire the flying lead end with a screw terminal block which will allow easy connection of the two wall switch wires.

Cheers,

alex

alphalove, 2016-11-25:
Hello,
Pull request sent with modified code.  Written and tested this only for the TH10.  Please make any changes you like.

Cheers,

alex

arendst, 2016-11-26:
Pull request implemented in version 2.0.18 just released

littlebilly, 2016-11-26:
Hi @arendst,
thanks a lot for implementation. May be one should change the topic,
as this enhancement is not only helpful for the HC-SR501 Human Infrared PIR Motion Sensor Detector
but for all switching sensors.

Cheers,
Billy

tam481, 2016-11-26:
Thanks a lot. Do you guys know if this could work with a standard Sonoff?

alphalove, 2016-11-27:
Hello tam481,
You could use the SCL pin that's presented on the J1 header (https://www.itead.cc/wiki/images/6/6b/Sonoff_schmatic.pdf).  This doesn't have an external pull-up so this would need to be initialised as INPUT_PULLUP.  You don't have the advantage of the 2.5mm connector for the break-out of the GPIO, but just solder on the leads and pass them though a hole in the casing.

**@arendst** - Thank you very much for pulling in the change!

Cheers,

alex

mihalski, 2016-11-27:
@tam481 You could follow these instructions to connect to the standard sonoff:

https://github.com/KmanOz/Sonoff-HomeAssistant#8-dht22-sensor-installation-for-original-sonoff-switch

General question with regards to usage with an external switch. The way I understand it, unless you add some logic between the ESP8266 and the switch, this will only work with momentary switches right?

arendst, 2016-11-27:
It also works with On/Off switches;
- SwitchMode 0 - Any switch change toggles relay
- SwitchMode 1 - Switch On: Relay On, Switch Off: Relay Off
- SwitchMode 2 - Switch On: Relay Off, Switch Off: Relay On

littlebilly, 2016-11-27:
To make it perfect one should have an additional SwitchMode 4 for alarm purposes for example.
  
  . SwitchMode 4 - Switch On: State published, Switch Off: State published

Depending on other circumstances the Relay On / Relay Off could be managed by the home automation system via MQTT.

Billy

arendst, 2016-11-27:
Is implemented in version 2.0.19 just released BUT it uses the already available ButtonTopic and ButtonRetain commands.

So to use it you'll need to define a topic other than "0" to ButtonTopic. If defined and an MQTT connection is present it will send ```cmnd/<ButtonTopic>/1/POWER (or LIGHT) ON (or OFF)```.

littlebilly, 2016-11-27:
Theo thank for the perfect implemantation in version 2.0.19.
Tested it with the standard Sonoff!

First, I set MQTT button topic to Alarm with  --> cmd ButtonTopic Alarm
--> Output
19:53:24 MQTT: tele/sonoff_2/TIME = 2016-11-27T19:53:24
19:54:35 MQTT: cmnd/Alarm/1/LIGHT = OFF

Runs perfectly.

Billy

tam481, 2016-11-28:
hey guys. Is his wiring correct:

http://i.imgur.com/Y0iZDPV.png

alphalove, 2016-11-30:
Hello tam481,
Umm, no that doesn't look correct.

The three connections are the following.
1. "AC Input" - Is exactly that.  Permanent AC supply.  The sonoff is always powered.
2. "Max Load 10A" output - This is connected to the light bulb.  The sonoff relay switches the active AC to the light when the relay is closed.
3. "SCL" & "GND" - The wall switch is connected to SCL and Sonoff GND pins (and nothing else!)

I'll mock up a sketch when I get home.  I'm a little confused as the J1 header in the photo only shows 4 holes, while the schematic has 5.  I just need to confirm that SCL is in fact present on that header.

Cheers,

alex

mihalski, 2016-11-30:
I'm with @alphalove on this. Also doing this sort of stuff in some countries (ie Australia) is highly illegal.

alphalove, 2016-11-30:
Hello mihalski,
Yep, it's a pity that CE mark does not equal C-Tick in Australia!

Cheers,

alex

mihalski, 2016-11-30:
And even if it did you'd still need an electrician to install it.

I'd also prefer better screw down terminals on the modules. And a better cutout that also works with round cables.

tam481, 2016-11-30:
hey guys. Many thanks for your help. I was confused as to how it would work. my switch only has 2 wires to the switch obviously live and the neutral going via the light bulb.
With regards to the SCL, my model does have it. I couldn't find a picture as good as the one I used to indicate where the connections would go.

@alphalove,
1. "AC Input" - there are two points to connect the wires to, as I understand it, one going through the relay and the other to the output point. The relay completes or disconnects the circuit as required. Right?

2. "Max Load 10A" output - Again the same question here, which of the two points?

I've created a diagram to illustrate my understanding and numbered the different pins. Please indicate what goes where by letter and number etc :-)

http://i.imgur.com/iBQS8Am.png

alphalove, 2016-11-30:
Hello tam481,
I had a look at a few of the old sonoff units I've got and they have a five pin header for J1, and SCL pin is broken out.  The very bad looking drawing i've attached should explain the required connections.

If your units don't have SCL broken out on J1, then you'll have to find another GPIO that is accessible for the switch input.  It might be possible to re-purpose the LED pin by  removing resistor R5 and connecting one leg of the wall switch to the R5 pad instead of the SCL pin.  Then modify the user_config.h to use the correct GPIO number for the input and put the LED output on an unused GPIO.

[image]

[image]

tam481, 2016-11-30:
Thank you so much @alphalove. My Sonoff does have the SCL header. Just figured out how to upload an image.

 I only have two wires. One live and the other neutral. Red and Black. The red is the live and goes into the switch which completes the circuit to the black via the light when switched on. That's why I was confused as to where wires would go.

[image]

closed 2016-12-02 by arendst