arendst/Sonoff-MQTT-OTA-Arduino #102: sonoff:62: error: 'TIME_T' has not been declared

namezero, 2016-11-25:
sonoff:62: error: 'TIME_T' has not been declared
sonoff:63: error: 'TIME_T' was not declared in this scope
sonoff:63: error: 'tm' was not declared in this scope
sonoff:64: error: 'TimeChangeRule' was not declared in this scope
sonoff:64: error: expected primary-expression before 'int'
sonoff:64: error: expression list treated as compound expression in initializer [-fpermissive]
sonoff:69: error: variable or field 'rtc_init' declared void
sonoff:69: error: 'rtcCallback' was not declared in this scope
sonoff:73: error: 'IPAddress' has not been declared
sonoff.ino: In function 'void setup()':
sonoff:2047: error: 'rtc_init' was not declared in this scope
support.ino: In function 'uint32_t makeTime(TIME_T&)':
support:491: error: 'uint32_t makeTime(TIME_T&)' redeclared as different kind of symbol
sonoff:63: error: previous declaration of 'uint32_t makeTime'
support.ino: In function 'uint32_t toTime_t(TimeChangeRule, int)':
support:522: error: 'uint32_t toTime_t(TimeChangeRule, int)' redeclared as different kind of symbol
sonoff:64: error: previous declaration of 'uint32_t toTime_t'
support:544: error: 'makeTime' cannot be used as a function
support.ino: In function 'void rtc_second()':
support:596: error: 'toTime_t' cannot be used as a function
support:597: error: 'toTime_t' cannot be used as a function
'TIME_T' has not been declared

cahbtexhuk, 2016-11-27:
Make sure you're using up-to-date Arduino IDE. I've spent few good days trying to compile this code with 1.6.5 and then automagically it compiled with 1.6.12 as suggested in prerequisites

0n3-70uch, 2016-11-29:
Yes, no problems here with arduino IDE 1.6.12.

Greetings
Fabi

closed 2016-11-29 by arendst

gnkarn, 2017-08-13:
im using arduino 1.8.1 IDE , with visualMicro , and the problem remains there, any advice?