arendst/Sonoff-MQTT-OTA-Arduino #106: Wall Switch not working

0n3-70uch, 2016-11-27:
Hello and thanks for this nice software :-).

I have a little issue with my sonoff: The wall switch function is not working. I tested the last firmware 2.0.19 and uncomment this:
 #define USE_WALL_SWITCH

Then I connected a cable to GND (Pin4) and GPIO14 (Pin5). But the relay is not switching...

Greetings
Fabian

arendst, 2016-11-27:
Pin4 Pin5 ? Apparently you are not using a TH10/16. In that case it might work if you change line 2090 in sonoff.ino from
```
  pinMode(SWITCH_PIN, INPUT); 
```
to
```
  pinMode(SWITCH_PIN, INPUT_PULLUP); 
```
Please give it a try and let me know.

0n3-70uch, 2016-11-27:
Oh thanks for your fast answere... but the problem still occur.

arendst, 2016-11-27:
What kind of sonoff are you using?

0n3-70uch, 2016-11-27:
Its a normal Sonoff:
https://www.itead.cc/sonoff-wifi-wireless-switch.html

arendst, 2016-11-27:
Does the switch on the sonoff itself work? What is the SwitchMode value?

0n3-70uch, 2016-11-27:
Yeah the build in switch is working and the MQTT command
`.../Power -m "toggle"`
is also working.

Where can I find the SwitchMode value?

arendst, 2016-11-27:
Execute command switchmode either using serial or via mqtt: cmnd/sonoff/switchmode

0n3-70uch, 2016-11-27:
Here is the log:
`19:19:46 MQTT: stat/Sonoff/SYNTAX = Status, SaveData, SaveSate, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Weblog, Syslog, LogHost, LogPort, SSId, Password, Hostname, Webserver
19:19:46 MQTT: stat/Sonoff/SYNTAX = MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, MqttUnits, MessageFormat, GroupTopic, Timezone, Light, Power, Ledstate, TelePeriod
19:19:46 MQTT: stat/Sonoff/SYNTAX = DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzUpdateTimer`

arendst, 2016-11-27:
But what returns command switchmode ?

If it returns only syntax then your loaded version does not contain wall switch functionality.

0n3-70uch, 2016-11-27:
Yeah this is the full log report after I send "switchmode" to the console.

Is there another place where I need to activate the switchmode?

arendst, 2016-11-27:
Nope. Are you sure you uploaded the correct compiled version?

arendst, 2016-11-27:
Just released 2.0.19a which also contains the SwitchMode command in the syntax message IF USE_WALL_SWITCH is enabled.

0n3-70uch, 2016-11-27:
I uploaded the code again... now I get this answer:

But the GPIO14 input is not working again. I tested 
`pinMode(SWITCH_PIN, INPUT);`
and
`pinMode(SWITCH_PIN, INPUT_PULLUP);`

0n3-70uch, 2016-11-27:
I found the error :-).

The Sonoff is demaged. I upload the sketch to another ESP and now it is working great with firmware 2.0.19a.

Thanks for your help...

closed 2016-11-27 by 0n3-70uch