arendst/Sonoff-MQTT-OTA-Arduino #107: Programm doesn't accept new MQTT_CLIENT_ID

0n3-70uch, 2016-11-27:
Hello again,

I changed the following in my user_config.h:
`#define MQTT_CLIENT_ID         "DVES_%06X"`
to
`#define MQTT_CLIENT_ID         "SONOFF_%06X"`

But it is not working. I found this line in sonoff.ino:
`#define DEF_MQTT_CLIENT_ID     "DVES_%06X"`

Is that the problem? Is it safe to change this line too?

Thank you
Fabian

arendst, 2016-11-27:
No.

0n3-70uch, 2016-11-27:
No = it's no problem to change the `DEF_MQTT_CLIENT_ID` or
No = thats a problem :-)

drhelmer, 2016-11-27:
I noted the same, was about to write a bug post, when I sawe you had posted this :)
 I changed the DEF_MQTT_CLIENT_ID in sonoff.ino, and, and got a new Fallback topic that I wanted. To me this seems like a bug, since the user_config.h should be enough I guess, dont know why it also should be in sonoff.ino, buty I didnt write the code :)

The problem now is that when you update the code, you need to overwrite this change all the time before uploading new firmware, so .....

Changing the CLIENT_ID in the webserver config MQTT menu doesnt work either.

0n3-70uch, 2016-11-27:
For me the this is working:
1) `#define MQTT_CLIENT_ID "YourName_%06X"`
2) `#define DEF_MQTT_CLIENT_ID "YourName_%06X"`
3) Update the firmware
4) Important point: Reset your device
...done

arendst, 2016-11-30:
Starting with version 2.0.20 you can change MQTTCLIENTID as you like but keep in mind that as soon as there is a percent sign used it will only allow a number following it converting that to the number of hexadecimal characters of the last digits of the MAC address as was originialy the case.

I have to do it this way as otherwise I had to provide a lot of code testing input values or the code would go haywire and the unit will panic or go in exception.

closed 2016-12-02 by arendst

0n3-70uch, 2016-12-05:
Works fine... Thank you.