arendst/Sonoff-MQTT-OTA-Arduino #108: MQTT blocking connection attempt

jaime1988, 2016-11-28:
Hi, first of all thanks for all your code and effort. 
It seems that when the device can not connect to the mqtt server, the button doesn't work correctly while the device is trying to connect. So if no correct mqtt is running, the device doesn't work correctly. Is there any possible solution?

0n3-70uch, 2016-11-29:
Hi, same result here. I think the problem also occurs if the device is not connected to a local wifi.

Greetings
Fabi

arendst, 2016-11-29:
What is worse without power the whole thing is useless!

0n3-70uch, 2016-11-29:
Yes you are right... a Sonoff without Wifi is just a power killer ;). 

There is only a little problem, if there is no Wifi (defect router) or no MQTT Broker (offline)... But for that it is ok when the button response is a little bit delayed.

Thanks for your good statement... for me this topic is closed :).

closed 2016-12-02 by arendst