arendst/Sonoff-MQTT-OTA-Arduino #109: Alexa

scargill, 2016-11-30:
Not an issue - a question... are you planning to incorporate the Alexa handling  in that pull request - that I think would be a great addition (you'd need a field for a meaningful device name "Alexa, turn the KITCHEN LAMP on" - where "KITCHEN LAMP" would be they unit name.  If you do this and handle multiple SSIDs (for moving units around the house!!)   I will happily immediately bin plans to extend my Sonoff code in favour of yours.

altelch, 2016-11-30:
The FakeWeMo-Code is optional. To have all features in one binary might be to much but if I need Alexa-Support but don't need TLS-MQTT, why not be able to select it. @scargill: Do you have an Echo device to test the code? You could build it from my fork just to see if it works. I could only test sending the UPnP M-Search and see the answer, then do the calls for the xml and see if they work. Thats exactly what Echo is supposed to do but without the real thing...

Greetings,

          Heiko.

scargill, 2016-11-30:
Yes I DO have 2 ECHO units - and am well familiar with them and would be more than happy to test code. (pete at scargill dot org if you want to save boring everyone until it is tested).

arendst, 2016-11-30:
Just released version 2.0.21 which contains the wemo code from Heiko.

I currently have no means to test it so I depend on Heiko.

Enable it in user_config.h as define USE_WEMO_EMULATION in line 75

alphalove, 2016-12-01:
Hello,
I have an Echo Dot.  I just tried the 2.0.21 version with USE_WEMO_EMULATION enabled.  Unfortunately the Echo does not find the device.

I use a 'wemo emu node' in node-red and this is discovered and works correctly  I think Mr. Scargill asked previously, were is the device name set?

Kind regards,

alex

altelch, 2016-12-01:
Hi Alex,

the Name is the variable MQTTClient. Just try to call 
http://<yoursonoffdevice>/setup.xml to see which name is used. This code 
is based on fauxmo but I couldn't really test it as I dont't own an 
Echo. Please set the LOGLEVEL to DEBUG and watch the log where you 
raised the level for. You should see a message regarding multicast join 
and if a search is coming in this is logged, too. If you won't get 
"Responding to search request ..." I'd need to see what request(s) are 
coming from the echo device. You could sniff the network from "outside" 
with wireshark or I could prepare a special debug version of my fork and 
you use that code for testing until this is resolved.

Greetings,

               Heiko.

alphalove, 2016-12-01:
Hello Heiko,
No worries.  Must the debug be a SYS_LOG host?  Or will web or serial log get the same details?  I don't have a syslog host setup, but can do if needed.  I'll fire wireshark up as well.

I see the xml file is using the DEVS_XXXXX name.

[image]

[image]

altelch, 2016-12-01:
On 01.12.2016 07:30, Alex Scott wrote:
>
> Hello Heiko,
> No worries. Must the debug be a SYS_LOG host? Or will web or serial 
> log get the same details? I don't have a syslog host setup, but can do 
> if needed. I'll fire wireshark up as well.
>
> I see the xml file is using the DEVS_XXXXX name.
>
Hi Scott,

web or serial is ok. Just did the same here. Changed mqtt device name to 
livingroom and now the setup.xml uses this as friendly name. I'd suspect 
the UDP part to be the problem.

Greetings,

                  Heiko.

altelch, 2016-12-01:
Hi Alex,

you might try this in support.ino line about 485:
.
.
.
      String request = packetBuffer;
+      addLog_P(LOG_LEVEL_DEBUG, packetBuffer);
      if(request.indexOf("M-SEARCH") > 0) {
.
.
.

Just to see what is in the UDP buffer.

Greetings,
                 Heiko.

arendst, 2016-12-01:
While at it try to change line 486 also in:
```
if (request.indexOf("-SEARCH") > 0) {
```
As the original code fails to detect the letter M as the first character of the string.

alphalove, 2016-12-01:
Hello arendst,
That fixed it!

Echo found the device ("Alexa, discover devices") and worked first go ("Alexa, turn study light on")

Great work!

Cheers,

alex

arendst, 2016-12-01:
Fixed and tuned in version 2.0.21a just released.

alphalove, 2016-12-01:
Hello Heiko,
Amazon don't sell the Echo device in Australia... But B&H Photo (https://www.bhphotovideo.com) do and have reasonable world wide shipping rates.

Thanks again for the code.

Cheers,
Alex

scargill, 2016-12-01:
Just catching up here - so does the main version now support Alexa?

altelch, 2016-12-02:
Hi Peter,

yes, Alex tested it successfully!

Greetings,

               Heiko.

Toshibass, 2016-12-02:
Think Im missing something simple here, were do I find/change :      mqtt device name

0n3-70uch, 2016-12-02:
Line 41 in user_config.h

`#define MQTT_CLIENT_ID         "DVES_%06X"  // Also fall back topic using Chip Id = last 6 characters of MAC address`

altelch, 2016-12-02:
Hi,

you can also change it in the web gui under mqtt settings or use mqtt to 
set it with

MqttClient. See 
https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/wiki/Commands.

Greetings,

                 Heiko.

scargill, 2016-12-02:
Right-  I'm getting a Sonoff out now to give it a whirl. Will report back.

scargill, 2016-12-02:
Well, I have to say (once I remembered the #define) it worked first time. Very nice. Will do more tests.

arendst, 2016-12-04:
Support for alternate SSID (AP) in version 2.1.2 just released

closed 2016-12-06 by arendst

dodacs, 2016-12-30:
Hi Guys, I am trying to compile the code but I am getting the errors below. Does anyone know what I am doing wrong? Thanks

Arduino: 1.6.7 (Linux), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

Board arduino:avr:uview doesn't define a 'build.board' preference. Auto-set to: AVR_UVIEW
/home/douglas/Arduino/ESP8266/sonoff/sonoff.ino: In function 'void CFG_Default()':
sonoff:487: error: 'addLog_P' was not declared in this scope
   addLog_P(LOG_LEVEL_NONE, PSTR("Config: Use default configuration"));
                                                                     ^
sonoff:489: error: 'CFG_Save' was not declared in this scope
   CFG_Save();
            ^
/home/douglas/Arduino/ESP8266/sonoff/sonoff.ino: In function 'void CFG_Migrate_Part2()':
sonoff:494: error: 'addLog_P' was not declared in this scope
   addLog_P(LOG_LEVEL_NONE, PSTR("Config: Migrating configuration"));
                                                                   ^
sonoff:600: error: 'CFG_Save' was not declared in this scope
   CFG_Save();
            ^
/home/douglas/Arduino/ESP8266/sonoff/sonoff.ino: In function 'void mqtt_publish(const char*, const char*, boolean)':
sonoff:758: error: 'addLog' was not declared in this scope
   addLog(LOG_LEVEL_INFO, log);
                             ^
/home/douglas/Arduino/ESP8266/sonoff/sonoff.ino: In function 'void mqtt_connected()':
sonoff:846: error: 'spiffsPresent' was not declared in this scope
     if (!spiffsPresent()) {
                        ^
/home/douglas/Arduino/ESP8266/sonoff/sonoff.ino: In function 'void mqtt_reconnect()':
sonoff:886: error: 'mqttDataCb' was not declared in this scope
     mqttClient.setCallback(mqttDataCb);
                            ^
sonoff:892: error: 'addLog_P' was not declared in this scope
   addLog_P(LOG_LEVEL_INFO, PSTR("MQTT: Attempting connection..."));
                                                                  ^
sonoff:907: error: 'addLog' was not declared in this scope
     addLog(LOG_LEVEL_DEBUG, log);
                                ^
/home/douglas/Arduino/ESP8266/sonoff/sonoff.ino: In function 'void mqttDataCb(char*, byte*, unsigned int)':
sonoff:925: error: 'addLog' was not declared in this scope
   addLog(LOG_LEVEL_DEBUG_MORE, svalue);
                                      ^
sonoff:1022: error: 'do_cmnd_power' was not declared in this scope
       do_cmnd_power(index, payload);
                                   ^
sonoff:1130: error: 'rtc_time' was not declared in this scope
             rtc_time(0).c_str(), rtc_time(1).c_str(), rtc_time(2).c_str(), rtc_time(3).c_str());
                       ^
sonoff:1133: error: 'rtc_time' was not declared in this scope
             rtc_time(0).c_str(), rtc_time(1).c_str(), rtc_time(2).c_str(), rtc_time(3).c_str());
                       ^
sonoff:1158: error: 'CFG_Save' was not declared in this scope
       CFG_Save();
                ^
sonoff:1267: error: 'WIFI_State' was not declared in this scope
         if (WIFI_State() != WIFI_RESTART) {
                        ^
sonoff:1430: error: 'send_button_power' was not declared in this scope
             send_button_power(i, 3);  // Clear MQTT retain in broker
                                   ^
sonoff:1460: error: 'addLog_P' was not declared in this scope
         addLog_P(LOG_LEVEL_INFO, PSTR("APP: Restarting"));
                                                         ^
sonoff:1495: error: 'CFG_Dump' was not declared in this scope
       CFG_Dump();
                ^
/home/douglas/Arduino/ESP8266/sonoff/sonoff.ino: In function 'void every_second()':
sonoff:1941: error: 'addLog_P' was not declared in this scope
         addLog_P(LOG_LEVEL_INFO, PSTR("SYSL: Syslog logging re-enabled"));  // Might trigger disable again (on purpose)
                                                                         ^
sonoff:2007: error: 'WIFI_getRSSIasQuality' was not declared in this scope
[... 203 lines cut ...]
webserver:370: error: 'handleWifi0' was not declared in this scope
     handleWifi0();
                 ^
/home/douglas/Arduino/ESP8266/sonoff/webserver.ino: In function 'void handleWifi1()':
webserver:529: error: 'handleWifi' was not declared in this scope
   handleWifi(true);
                  ^
/home/douglas/Arduino/ESP8266/sonoff/webserver.ino: In function 'void handleWifi0()':
webserver:534: error: 'handleWifi' was not declared in this scope
   handleWifi(false);
                   ^
/home/douglas/Arduino/ESP8266/sonoff/webserver.ino: In function 'void handleNotFound()':
webserver:1177: error: 'captivePortal' was not declared in this scope
   if (captivePortal()) { // If captive portal redirect instead of displaying the error page.
                     ^
/home/douglas/Arduino/ESP8266/sonoff/webserver.ino: In function 'boolean captivePortal()':
webserver:1201: error: 'isIp' was not declared in this scope
   if (!isIp(webServer->hostHeader())) {
                                    ^
exit status 1
'addLog_P' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

arendst, 2016-12-30:
Dodacs, you're doing wrong many things:

1. This is a closed issue. Expect no response
2. This issue is about Alexa, not compile errors
3. You're using an unsupported version of the Arduino IDE (1.6.7).
4. You should have read the wiki and set-up an environment as mentioned on the prerequisite page.

Better luck next time ;-)

dodacs, 2016-12-30:
Sorry for posting on the wrong place and thank you for your reply. Updating the IDE fixed the problem! 
You have done a great job. Very nice update to the sonoff. Keep up the good work!

Regards,