arendst/Sonoff-MQTT-OTA-Arduino #111: Home Assistant Mosquitto Integration Issue

prahjister, 2016-12-02:
I am having some major issues with getting integration to work and I do not know where to start troublehoot.

I am running Home Assistant and Mosquitto on a Windows 10 box. Both appear to be the latest and installed ok.

Here is my sonoff config.

```
light:
  platform: mqtt
  name: "Living Room Light"
  state_topic: "stat/sonoff/POWER"
  command_topic: "cmnd/sonoff/power"
  payload_on: "ON"
  payload_off: "OFF"
```
and my mqtt config

```
mqtt:
  broker: 10.0.0.200
  port: 1883
  client_id: home-assistant-1
```

sonoff device config

[image]

When I compiled and uploaded the only thing i changed was my ssid and password.

Here are my symptoms.

I can turn the relay on and off fine from an app on my phone called mymqtt. It successfully connects. And if I publish a 1 or 0 to cmnd/sonoff/power from the app.

When I press the button on the sonoff the relay works. but neither the mymqtt app shows status nor home assistant.

On the Home Assistant states page if the light is on I can press the toggle and it turns off. I can see the on request on the mymqtt app. If I try to turn it on with the toggle sonoff blinks but does nothing.

Any help is appreciated. I have spent a lot of time troubleshooing. Did a lot of reading but this is as far as i have got.

arendst, 2016-12-02:
What happens if you change ```state_topic``` in ```stat/sonoff/1/POWER```

prahjister, 2016-12-02:
I am not positive where you want me to change...in the arduino code? I did try in the home assistance config and now no state change even with phone.

arendst, 2016-12-02:
In your home assistance config.

Just try a basic mosquitto command like  ```mosquitto_pub -t cmnd/sonoff/POWER -m 2```
 and monitor with  ```mosquitto_sub -t stat/# -t tele/# -t cmnd/# -v```

prahjister, 2016-12-02:
Seems to work. I have an output and everything works as expected. But when i pressButton on sonoff only show stat and no cmnd when pressing the button on the device. Phone works as expected. I did put home assistance config back. 

tat/sonoff/1/POWER OFF
cmnd/sonoff/power 1
stat/sonoff/POWER ON
cmnd/sonoff/power 2
stat/sonoff/POWER OFF
cmnd/sonoff/power 0
stat/sonoff/POWER OFF
cmnd/sonoff/power 2
stat/sonoff/POWER ON
stat/sonoff/1/POWER OFF
stat/sonoff/1/POWER ON
stat/sonoff/1/POWER OFF

prahjister, 2016-12-02:
i see....it is mixing stat

arendst, 2016-12-02:
Pressing the button is not a command so you won't see a ```cmnd``` UNLESS you configured ButtonTopic in which case the button will send a MQTT command and you have to act on it. 

Forget about ButtonTopic now and focus on the return stat value. The sonoff software works as designed. Your Home assistance script needs some tweeking.

prahjister, 2016-12-02:
i was under the impression that i could use the button on the sonoff to turn the relay on and off.

prahjister, 2016-12-02:
and why does it return status on stat/sonoff/1/.......is there a way i can modify to remove the 1

arendst, 2016-12-02:
Well you do.

Start from the beginning (ButtonTopic = 0):
- Press button and relay toggles - Mqtt response: ```stat/sonoff/1/POWER ON```
- Press button and relay toggles - Mqtt response: ```stat/sonoff/1/POWER OFF```

And no you can't remove the one. It is needed in sonoff dual and future devices to distinguish between relays.

arendst, 2016-12-02:
Just send mqtt commands with the one too like ```cmnd/sonoff/1/POWER 1``` and all is well.

prahjister, 2016-12-02:
Almost 100% I really appreciate your help....Button now works on the sonoff...
only thing that doesn't seem to work as I expect then is when I send an "ON" it doesn't turn on. Only with numbers.

arendst, 2016-12-02:
What happens with command ``` mosquitto_pub -t cmnd/sonoff/POWER -m ON```

Mine works.

arendst, 2016-12-02:
Ah you're right with the 1 in it it doesn't. I'll investigate...

arendst, 2016-12-02:
Solved in next version soon to be released.

prahjister, 2016-12-02:
Awesome...once confirmed I will post working config for newbies like me to this. Hopefully save them some frustration.

arendst, 2016-12-02:
New version released. Please confirm.

prahjister, 2016-12-02:
Perfect...so here is the final config. Now to figure out Home Assistant.....I wish there was an easier way to automate...

```
light:
  platform: mqtt
  name: "Living Room Light"
  state_topic: "stat/sonoff/1/POWER"
  command_topic: "cmnd/sonoff/1/power"
  payload_on: "ON"
  payload_off: "OFF"
```

```
mqtt:
  broker: 10.0.0.200
  port: 1883
  client_id: home-assistant-1
```
[image]

closed 2016-12-02 by arendst