arendst/Sonoff-MQTT-OTA-Arduino #117: Extra pins on Sonoff Dual

scientistnobee, 2016-12-05:
Dear Arendst,

I would like to add a temperature sensor to Dual. I can see some extra pins on Dual's PCB. Do know what are these for?

arendst, 2016-12-05:
See web page iTead: [https://www.itead.cc/smart-home/sonoff-dual.html](https://www.itead.cc/smart-home/sonoff-dual.html)

scientistnobee, 2016-12-05:
Thanks for your quick reply. But I can't see any information on the webpage about connectors: (VCC, C2K, C2D, GND) and (VCC, BUTTON 0, BUTTON 1, GND). That is why I am asking you.

arendst, 2016-12-05:
Look harder. Down the page is a link to the Dual datasheet where you find among other stuff the schematics. Other than those pin names you won't find nothing usefull I'm afraid.

scientistnobee, 2016-12-05:
Thanks again for your help. I posted in the support forum.
http://support.iteadstudio.com/support/discussions/topics/11000006870
Hope someone will reply.

altelch, 2016-12-06:
Hi, as arendst statet, use the schematics. If the pin has no name in the schematics, there is no connection somewhere on the board. Itead is very economical with available connections. Unfortunately with the dual none of the "standard" GPIOs 4/14 are available as with the Sonoff TH. You need to solder them directly to the ESP8266 what isn't easy...

scientistnobee, 2016-12-06:
Thanks altelch. I will try to directly solder to pin 6, something like shown in http://tinkerman.cat/your-laundry-is-done/

closed 2016-12-27 by arendst

ageorgios, 2017-03-27:
I need two extra GPIO Pins on Sonoff Dual. Can you advise what to solder? or what is the best way to achieve two GPIOs? On Tasmota Firmware when configured in Sonoff Dual it has GPIO4 available.

mydakota, 2017-07-19:
ageorgios Id like to know the answer to your question too.
if its possible to have like the single sonoff acting as a light switch with switch mode 3 set for a lights switch works well on standard sonoff but I cant see a way of actuality controlling two lights in the same way as seems to only allow 1 single input please advise if its possible to controll two lights and interface two light switches using GPIO pins thanks