arendst/Sonoff-MQTT-OTA-Arduino #119: Extra GPIO for all Sonoff?

Reinhartd, 2016-12-06:
Hello Theo!

can you make a switch to disable the serial communication (RxD + TxD), flashing is everyone over OTA.
Then we can use the gpio1 and gpio3 for other devices on the output pins!

LG

arendst, 2016-12-06:
Won't work for the dual as it uses serial to communicate with chip controlling the relays.

Reinhartd, 2016-12-06:
The Dual and Slampher is a exception, but the normal Switch or Pow are predestined for more GPIO's!
I think, DS18b20 or DHT22 is very good and your Sketch can work with this devices.

LG

dagbdagb, 2016-12-06:
Would this also require code to somehow re-enable RxD/TxD if the need should arise?

Reinhartd, 2016-12-06:
a good solution is a timeout ( 1- 2 min) at power on. After 2 minutes the serial communication is automatic disabled, then we can flashing the device witout OTA.  Another idea is a switch in the Webif. 

LG

arendst, 2016-12-07:
I would NOT consider the pow for (external) sensors:
- gnd is connected to AC which might be 240V and could harm you(r) kids
- energy monitoring  is very interrupt driven; the more power you use the more interrupts need to be serviced

You should start using seperate sensor nodes using esp-easy and hook them up with a home automation tool like  mqtt, node-red, domoticz, openhab, fhem ...

Apart from that I will consider disabling serial in a controlled fashion.

stoehrmark, 2016-12-17:
I have been using this code with a DHT22 connected to GPIO3 TX. 
No problem just change:
#define DHT_PIN              14  
to
#define DHT_PIN              3
in the  user_config.h no other changes.

closed 2016-12-27 by arendst