arendst/Sonoff-MQTT-OTA-Arduino #120: Toggle switch on Configuration URL using MQTT topic/message?

jaquesp, 2016-12-06:
This has confused me a little this evening.
Does the toggle switch on the configuration screen actually publish an MQTT message to toggle the switch on and off? If I separately subscribe to the topic I can't see the messages coming through.

Thanks in advance,
Paul

arendst, 2016-12-06:
It does NOT send an MQTT message. It does switch the relay directly which will then send a status message.

Only the button on the device can be configured to send a MQTT message with command ButtonTopic.

jaquesp, 2016-12-06:
I just tried to do that with the following commands, but the button doesn't seem to be sending messages:

`mosquitto_sub -u evsefgcx -P U3sjB4MtVLvH -t cmnd/sonoff/POWER
mosquitto_pub -u evsefgcx -P U3sjB4MtVLvH -t cmnd/sonoff/ButtonTopic -m "sonoff"
mosquitto_pub -u evsefgcx -P U3sjB4MtVLvH -t cmnd/sonoff/ButtonRetain -m "off"`

Any advice gratefully received.

Paul

arendst, 2016-12-06:
If you use command ```mosquitto_sub -t cmnd/sonoff/# -v``` to subscribe you'll see that a button press always includes a relay index. This is confusing and I'm in the process of eliminating this.

But for now you'll have to deal with it.

So pls use commands like ```cmnd/sonoff/1/power on``` to be consistent.

jaquesp, 2016-12-06:
Understood - that makes perfect sense. I amended the homebridge configuration to use the relay index and everything now works as expected! Thanks for your help!

closed 2016-12-06 by jaquesp