arendst/Sonoff-MQTT-OTA-Arduino #121: Can I flash Wemos D1 mini to test my setup?

yaydogan, 2016-12-07:
I am waiting for Sonoff switch. Can I flash my Wemos D1 mini to test it out in my home setup?

Thanks...

arendst, 2016-12-07:
Give it a try. What could happen? Let us know.

Btw, among others I develop on wemos1 d1 mini ...

climberhunt, 2016-12-07:
It does work, but with a couple tweaks. 
The check in sonoff.ino for checking the value of REL_PIN at boot results in it being recognized as a SONOFF_DUAL, I force the syscfg.model to 1 at this stage. Also, I changed the GPIO pins in the cfg to use the 'D' numbers for convenience. Works well.

arendst, 2016-12-08:
It does work as a Sonoff if you connect a relay to the correct GPIO. If not perform the undocumented command ```model 1```.

yaydogan, 2016-12-08:
Thank you for your feedback!

closed 2016-12-08 by arendst

rlust, 2016-12-23:
I am trying the Wemos D1 with a relay shield. What GPIO did you use to get the relay to work?
I have a second one working using the DHT11 shield and I am getting temp readings.

arendst, 2016-12-23:
I'm emulating Sonoff so the relay is connected to GPIO12. See user_config.h for further information