arendst/Sonoff-MQTT-OTA-Arduino #122: Thoughts on using GPIO14 as a duplicate for the standard button

Malarcy, 2016-12-07:
Hi - this is a question - have just received a couple of sonoff devices, appear to the newer 5 pin/no hold for the led in the cover ones. From reading I think they are the TH versions.

I see in the code options around temp sensors and the like, but what I would like to do is to hook up this pin (GPIO14) to an external switch (in my case a floor switch - but I can see other use cases) - and have it do the same as the standard switch (GPIO0) - I want local control of the relay ( for the family) but they won't be accessing the core module (that's going to be inside/behind a unit). I also want to retain the multi-press options 

So - rather than attempting to wire in parallel to the existing switch, my thoughts were to duplicate the function of KEY_PIN (or even just swap KEY_PIN to 14) to provide remote only

Interested in thoughts about this, or any other approach to achieving a local control button that can be made "family friendly"- perhaps a wall switch, perhaps a foot switch.

alphalove, 2016-12-07:
Hello Malarcy,
This functionality was in release 2.0.18.  You can have a read of the details here.

https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/issues/101

Cheers,

alex

Malarcy, 2016-12-07:
Yes - that covers it - sounds exactly what I was looking for - sorry for the duplicate and thanks for the response.

closed 2016-12-07 by Malarcy