arendst/Sonoff-MQTT-OTA-Arduino #123: measuring 0V while no load connected?

kwittwer, 2016-12-07:
Hi volks, 
I'm getting all zeros for voltage current an power while no or small loads are connected (5W). Is this normal? Shouldn't the device show up a voltage even with no connected load? (the relay is closed -> on)
thanks for your answers!

kwittwer, 2016-12-07:
[image]

arendst, 2016-12-07:
If relay is off I decided to zero our all values. Although the sensor is measuring before the relay I thought it doesn't make sense to provide values when the relay is off.

The problem of no or intermittent values at loads below 6W is also as expected. With such low loads small amount of interrupts are received leading to inconsequent power values.

Toshibass, 2016-12-10:
I agree with first statement 0v is fine when the relay is open BUT I strongly disagree with the second statement, I am still using **version 2.1.10** for **Sonoff_pow** because its stable and consistent at loads below 4w, I don't get any zero or intermittent values at all, however when I have tried updating to any later version higher than 2.1.10 then I get the intermittent values returned which is no good for me.

arendst, 2016-12-10:
Will look into it inthe near future.

arendst, 2016-12-17:
I've updated the power measurement to better support low power situations in version 3.0.7 just released.

It's a matter of decision when to detect no power or low power. It appears that a 4W load takes almost 3 seconds to measure (interrupts are generated every three seconds for 4W, higher loads generate more interrupts). In the previous releases the decision point was at 1 second leading to what you noticed..

So to find a nice equilibrium I now decided that when it takes more than three seconds to receive an interrupt no load is present and 0 values will be returned for power, voltage and current.

Give it a go and let me know.

arendst, 2016-12-18:
Again an update in version 3.0.9 just released. This time the voltage reading will stay up when the relay is on but no load is present as requested and is also a safeguard.

kwittwer, 2016-12-18:
Hey, sounds great I will try it :-)

kwittwer, 2016-12-20:
I've tried it and it works like a charm! thanks!

closed 2016-12-21 by arendst

Toshibass, 2016-12-22:
Sorry I have been on vacation, so only just found time to test 3.1.0 at works great thanks for all your time and effort.

paveleremin, 2016-12-23:
@arendst @kwittwer @Toshibass hello guys!
I'm want to make some notification via OpenHAB when the wash machine is done. I plan use Sonoff POW. Is it possible?
Can someone share some help to understand how Sonoff POW work with custom firmware? I'm talking about mesurement, not switch functionality.

arendst, 2016-12-23:
I would say give it a try and let us know.