arendst/Sonoff-MQTT-OTA-Arduino #126: Retain flag on stat message

Luc3as, 2016-12-07:
Hello, I would need to add support for optionally set true to retain flag on stat messages.
I have following problem in use with HomeAssistant, I am able to turn on and off two way from home assistant to sonoff, I can show actual status, but problem is, when I restart home assistant, it is looking for status message at  stat/sonoff/1/POWER ON   but cannot find that message because it  has no retain flag, and thus the initial state of component in Homeassistant is off, even if sonoff is on. 
If I turn sonoff off and on, the status is correct again but I think  if there would be chance to send stat message with retain flag , this problem would be solved.

I tried to look into the code, but found 2 definition of mqtt_publish , one with retain option and second without, I dont know how they are ment and If I could change publish of stat message with additional parameter in function call..

`void mqtt_publish(const char* topic, const char* data, boolean retained)
{
  char log[TOPSZ+MESSZ];

  if (mqttClient.publish(topic, data, retained)) {
    snprintf_P(log, sizeof(log), PSTR("MQTT: %s = %s%s"), topic, data, (retained) ? " (retained)" : "");
//    snprintf_P(log, sizeof(log), PSTR("MQTT: %s = %s%s"), strchr(topic,'/')+1, data, (retained) ? " (retained)" : ""); // Skip topic prefix
//    mqttClient.loop();  // Do not use here! Will block previous publishes
  } else {
    snprintf_P(log, sizeof(log), PSTR("RSLT: %s = %s"), topic, data);
  }
  addLog(LOG_LEVEL_INFO, log);
  blinks++;
}

void mqtt_publish(const char* topic, const char* data)
{
  mqtt_publish(topic, data, false);
}`

arendst, 2016-12-07:
Why don't you use Domoticz? It's a lot more intelligent as it keeps the last state although I can't remember I had to stop Domoticz...

Luc3as, 2016-12-07:
Because I have home automation built on HomeAssistant and I won't migrate all system to something other now.

timtimsson, 2016-12-07:
... i also use openhab and retrain and lwt are very usefull things and would improve the app ...

arendst, 2016-12-08:
I'll implement in next version.

arendst, 2016-12-08:
Well guys it's implemented in version 3.0.0 just released and can be enabled with ```cmnd/sonoff/PowerRetain On``` but pay attention:

The overall process of controlling sonoff has changed to make it more consistent. So if you used to switch a relay using the number in between like ```cmnd/sonoff/1/power on``` you should now use ```cmnd/sonoff/power on```. The response will now always be ```stat/cmnd/POWER ON``` with optional JSON message if enabled.

For the Dual and Dragon the relays are addressed using POWER1 and POWER2.

See the wiki for updated commands and MQTT Features.

Let me know if it works for you.

Luc3as, 2016-12-08:
Working like a charm. Many thanks for support and quick implementation.

closed 2016-12-08 by Luc3as