arendst/Sonoff-MQTT-OTA-Arduino #134: On off switch with timer

stoehrmark, 2016-12-11:
Would it be possible to add a new command (Light 3, Light PUSHBUTTON) where the relay turns ON then a time delay then OFF. It could be used to trigger something with the relay. EG Garage Door or door maglock. Thanks and Excellent work on this!!

e.g.
user_config.h
#define DoorActivationPeriod  600                            // [ms] Duration to close the switch.

sonoff.ino
    digitalWrite(LED, LOW);
    digitalWrite(REL_PIN, HIGH);
    delay(DoorActivationPeriod);
    digitalWrite(LED, HIGH);
    digitalWrite(REL_PIN, LOW);

stoehrmark, 2016-12-11:
I am using this for a Garage door controller node. I used a hacksaw on the sonoff to disconnect the relay form the Mains AC and make it stand alone. In the very basic script I have GPIO14 is attached to a reed switch and used to indicate True False if the door is open. I used the Serial RX line GPIO3 connected to a DHT22 to read temp and Humidity. That works fine, leaving the TX GPIO1 for Serial communications back to the computer for testing. (nothing fancy just  #define DHTPIN  3   //Using TX pin GPIO3)

Is there a way to use GPIO14 as a True False indication.

[image]

arendst, 2016-12-15:
Will investigate

arendst, 2016-12-17:
In version 3.0.6 just released you have the option of using command PulseTime to enter a value * 0.1 Sec to keep the relay on after which it will turn off. In your case you would enter 6 to get 600 mSec. Setting it to 0 disbales this feature. Also see commands in wiki.

So after turning the relay On with any power on command will turn it off after 600 mSec.

stoehrmark, 2016-12-17:
Will give it a go thanks

closed 2016-12-27 by arendst