arendst/Sonoff-MQTT-OTA-Arduino #135: Sonoff tries connecting to indebuurt

legolas, 2016-12-11:
Despite that I did change the SSID1 and 2 inside the user_config.h I still get this output:

0:00:00 APP: Project sonoff (Topic sonoff, Fallback DVES_073FFB, GroupTopic sonoffs) Version 3.0.3
00:00:00 Wifi: Connecting to AP2 indebuurt3 (VnsqrtnrsddbrN) in mode 11N as sonoff-8187...
00:00:07 Wifi: Connect failed as AP cannot be reached
00:00:07 Wifi: Connecting to AP1 indebuurt2 (VnsqrtnrsddbrN) in mode 11N as sonoff-8187...
00:00:14 Wifi: Connect failed as AP cannot be reached
00:00:14 WPSconfig: Active for 1 minute

Obviously the connection fails.
What am I doing wrong here?

arendst, 2016-12-11:
Use commands ssid1 and password1 to change to your wifi network. user_config.h is only used once during initial install.

Alternatively change define CFG_HOLDER.

legolas, 2016-12-12:
Thanks for the reply.
The above log is from the first boot after install. I would've expected it to take my changes.

legolas, 2016-12-13:
Good morning Theo,

Is it sufficient to change the "Date" (at least it looks like a date to me) value of CFG_HOLDER to a newer date to force reloading the defaults as defined in user_config.h?

arendst, 2016-12-13:
Marcel, yep.

legolas, 2016-12-13:
Thanks, will try that tonight when I'm home.

legolas, 2016-12-15:
Tonight I finally had the time to test this and it works! Thank for the help.
Next step: control the sonoff using openhab.

closed 2016-12-15 by legolas

weerada-moria, 2017-09-18:
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_92D241, GroupTopic sonoffs) Version 5.7.1i
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as sonoff-4673...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 indebuurt1 in mode 11N as sonoff-4673...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfigactive for 1 minute

i found the same problem as your  how could you solve it ? i don't have any luck with this

legolas, 2017-09-18:
@weerada-moria I would suggest switching to https://github.com/arendst/Sonoff-Tasmota