arendst/Sonoff-MQTT-OTA-Arduino #136: Add option to supress blinks for telemetry/timer events

thomasf, 2016-12-11:
I'm not sure exactly which events I would like to supress LED blinking for but I think it's good for the telemetry messages only..

Maybe something like this (?:

```diff
diff --git a/sonoff/sonoff.ino b/sonoff/sonoff.ino
index c26d39f..766cdd5 100644
--- a/sonoff/sonoff.ino
+++ b/sonoff/sonoff.ino
@@ -1934,6 +1934,9 @@ void every_second()
         snprintf_P(svalue, sizeof(svalue), PSTR("%s, \"Wifi\":{\"AP\":%d, \"SSID\":\"%s\", \"RSSI\":%d}}"),
           svalue, sysCfg.sta_active +1, sysCfg.sta_ssid[sysCfg.sta_active], WIFI_getRSSIasQuality(WiFi.RSSI()));
         mqtt_publish(stopic, svalue);
+#ifdef TELE_LED
+        blinks--;
+#endif
       } else {
 #ifdef SEND_TELEMETRY_POWER
         for (i = 0; i < Maxdevice; i++) {
@@ -1944,23 +1947,30 @@ void every_second()
           }
           strlcpy(svalue, (power & (0x01 << i)) ? MQTT_STATUS_ON : MQTT_STATUS_OFF, sizeof(svalue));
           mqtt_publish(stopic, svalue);
+#ifdef TELE_LED
+          blinks--;
+#endif
         }
 #endif  // SEND_TELEMETRY_POWER
 #ifdef SEND_TELEMETRY_UPTIME
         snprintf_P(stopic, sizeof(stopic), PSTR("%s/%s/UPTIME"), PUB_PREFIX2, sysCfg.mqtt_topic);
         snprintf_P(svalue, sizeof(svalue), PSTR("%d%s"), uptime, (sysCfg.mqtt_units) ? " Hr" : "");
         mqtt_publish(stopic, svalue);
+        blinks--;
 #endif  // SEND_TELEMETRY_UPTIME
 #ifdef SEND_TELEMETRY_WIFI
         snprintf_P(stopic, sizeof(stopic), PSTR("%s/%s/AP"), PUB_PREFIX2, sysCfg.mqtt_topic);
         snprintf_P(svalue, sizeof(svalue), PSTR("%d"), sysCfg.sta_active +1);
         mqtt_publish(stopic, svalue);
+        blinks--;
         snprintf_P(stopic, sizeof(stopic), PSTR("%s/%s/SSID"), PUB_PREFIX2, sysCfg.mqtt_topic);
         snprintf_P(svalue, sizeof(svalue), PSTR("%s"), sysCfg.sta_ssid[sysCfg.sta_active]);
         mqtt_publish(stopic, svalue);
+        blinks--;
         snprintf_P(stopic, sizeof(stopic), PSTR("%s/%s/RSSI"), PUB_PREFIX2, sysCfg.mqtt_topic);
         snprintf_P(svalue, sizeof(svalue), PSTR("%d%s"), WIFI_getRSSIasQuality(WiFi.RSSI()), (sysCfg.mqtt_units) ? " %" : "");
         mqtt_publish(stopic, svalue);
+        blinks--;
 #endif  // SEND_TELEMETRY_WIFI
       }
 
@@ -2099,27 +2109,35 @@ void every_second()
         snprintf_P(svalue, sizeof(svalue), PSTR("{\"Time\":\"%s\", \"Energy\":{\"Yesterday\":\"%s\", \"Today\":\"%s\", \"Period\":%d, \"Power\":%d, \"Factor\":\"%s\", \"Voltage\":%d, \"Current\":\"%s\"}}"),
           stime, stemp0, stemp1, pe, pw, stemp2, pu, stemp3);
         mqtt_publish(stopic, svalue);
+        blinks--;
       } else {
         snprintf_P(stopic, sizeof(stopic), PSTR("%s/%s/YESTERDAY_ENERGY"), PUB_PREFIX2, sysCfg.mqtt_topic);
         snprintf_P(svalue, sizeof(svalue), PSTR("%s%s"), stemp0, (sysCfg.mqtt_units) ? " kWh" : "");
         mqtt_publish(stopic, svalue);
+        blinks--;
         snprintf_P(stopic, sizeof(stopic), PSTR("%s/%s/TODAY_ENERGY"), PUB_PREFIX2, sysCfg.mqtt_topic);
         snprintf_P(svalue, sizeof(svalue), PSTR("%s%s"), stemp1, (sysCfg.mqtt_units) ? " kWh" : "");
         mqtt_publish(stopic, svalue);
+        blinks--;
         snprintf_P(stopic, sizeof(stopic), PSTR("%s/%s/PERIOD_ENERGY"), PUB_PREFIX2, sysCfg.mqtt_topic);
         snprintf_P(svalue, sizeof(svalue), PSTR("%d%s"), pe, (sysCfg.mqtt_units) ? " Wh" : "");
         mqtt_publish(stopic, svalue);
+        blinks--;
         snprintf_P(stopic, sizeof(stopic), PSTR("%s/%s/CURRENT_POWER"), PUB_PREFIX2, sysCfg.mqtt_topic);
         snprintf_P(svalue, sizeof(svalue), PSTR("%d%s"), pw, (sysCfg.mqtt_units) ? " W" : "");
         mqtt_publish(stopic, svalue);
+        blinks--;
         snprintf_P(stopic, sizeof(stopic), PSTR("%s/%s/POWER_FACTOR"), PUB_PREFIX2, sysCfg.mqtt_topic);
         mqtt_publish(stopic, stemp2);
+        blinks--;
         snprintf_P(stopic, sizeof(stopic), PSTR("%s/%s/VOLTAGE"), PUB_PREFIX2, sysCfg.mqtt_topic);
         snprintf_P(svalue, sizeof(svalue), PSTR("%d%s"), pu, (sysCfg.mqtt_units) ? " V" : "");
         mqtt_publish(stopic, svalue);
+        blinks--;
         snprintf_P(stopic, sizeof(stopic), PSTR("%s/%s/CURRENT"), PUB_PREFIX2, sysCfg.mqtt_topic);
         snprintf_P(svalue, sizeof(svalue), PSTR("%s%s"), stemp3, (sysCfg.mqtt_units) ? " A" : "");
         mqtt_publish(stopic, svalue);
+        blinks--;
       }
 #endif  // SEND_TELEMETRY_ENERGY
 #endif  // USE_POWERMONITOR
@@ -2128,6 +2146,7 @@ void every_second()
         snprintf_P(stopic, sizeof(stopic), PSTR("%s/%s/TIME"), PUB_PREFIX2, sysCfg.mqtt_topic);
         snprintf_P(svalue, sizeof(svalue), PSTR("%s"), stime);
         mqtt_publish(stopic, svalue);
+        blinks--;
       }
     }
   }
diff --git a/sonoff/user_config.h b/sonoff/user_config.h
index 3944017..d5ee298 100644
--- a/sonoff/user_config.h
+++ b/sonoff/user_config.h
@@ -72,6 +72,7 @@
 
 // -- MQTT - Telemetry -----------------------
 #define TELE_PERIOD            300          // [TelePeriod] Telemetry (0 = disable, 10 - 3600 seconds)
+#define TELE_LED                            // Comment to disable LED blinking on telemetry timer
 #define SEND_TELEMETRY_UPTIME               // Enable sending uptime telemetry (if disabled will still send hourly message)
 #define SEND_TELEMETRY_WIFI                 // Enable sending wifi telemetry
 #define SEND_TELEMETRY_POWER                // Enable sending power telemetry

```

arendst, 2016-12-15:
Will implement more control over LED

arendst, 2016-12-15:
Several options added to command LedState allowing more control over it's use in version 3.0.5 just released.

See wiki or _releasenotes.ino

closed 2016-12-17 by arendst