arendst/Sonoff-MQTT-OTA-Arduino #137: Identify telemetry sending device

guglielmino, 2016-12-12:
In a condition where You have many pow devices it may be useful to know which one is sending telemetry data. Doing some tries with current firmware I see all devices publish telemetry data on the same mqtt topic, so it's not possibile to know who is sending the data. Would it not be better to add device name inside the telemetry payload?

thomasf, 2016-12-12:
That's not how the code looks to me, (I don't have a POW device and that I just skimmed through that code).

thomasf, 2016-12-12:
And all basic telemetry data is correctly sent on the individual device topic of the device..

ie.. I renamed a default confirued one and it's telemetry jumped topic:

from

```
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/sonoff/POWER', ... (3 bytes))
OFF
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/sonoff/UPTIME', ... (1 bytes))
0
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/sonoff/AP', ... (1 bytes))
1
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/sonoff/SSID', ... (4 bytes))
testfi
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/sonoff/RSSI', ... (2 bytes))
56
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/sonoff/TIME', ... (19 bytes))
2016-12-12T10:02:27
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/sonoff/UPTIME', ... (1 bytes))
1
```

to

```
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/TESTY/POWER', ... (3 bytes))
OFF
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/TESTY/UPTIME', ... (1 bytes))
1
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/TESTY/AP', ... (1 bytes))
1
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/TESTY/SSID', ... (4 bytes))
testfi
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/TESTY/RSSI', ... (2 bytes))
52
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/TESTY/TIME', ... (19 bytes))
2016-12-12T10:04:07
Client mosqsub/26101-transwhal received PUBLISH (d0, q0, r0, m0, 'stat/TESTY/INFO', ... (74 bytes))
Sonoff module version 3.0.4, FallbackTopic DVES_06E2F6, GroupTopic sonoffs

```

guglielmino, 2016-12-12:
Before trying I looked at the code and, as You can see, TELEMETRY data is sent on default topic, below data received by a test program I done:

```json
   {  
         "Time":"2016-12-10T14:28:37",
         "Energy":{  
            "Yesterday":"0.000",
            "Today":"0.000",
            "Period":0,
            "Power":0,
            "Factor":"0.00",
            "Voltage":0,
            "Current":"0.000"
         }
    }
```
This message was received in a configuration with two sonoff pow, they had different fallback topic (as obvious) but published on the same topic (tele/sonoff/TELEMETRY). This make sense for me but I think it would be better to have something like:

```json
   {  
         "DeviceName": "DVES_20D9AC",
         "Time":"2016-12-10T14:28:37",
         "Energy":{  
            "Yesterday":"0.000",
            "Today":"0.000",
            "Period":0,
            "Power":0,
            "Factor":"0.00",
            "Voltage":0,
            "Current":"0.000"
         }
    }
```

thomasf, 2016-12-12:
Ah.,. I reread the code and it seems like this problem related to `JSON` format only which I don't use.

thomasf, 2016-12-12:
But... The host name should not be used as a device identifier, you give each device it's own topic and identify them that way. Also, you can't address/send commands to your sonoffs individually if you don't do this.

When I changed telemetry to JSON format I still got the message under my TESTY name:
```
Client mosqsub/30919-transwhal received PUBLISH (d0, q0, r0, m0, 'tele/TESTY/TELEMETRY', ... (100 bytes))
{"Time":"2016-12-12T11:07:58", "Uptime":0, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"testfi", "RSSI":58}}

```

guglielmino, 2016-12-12:
As far as I understood every device has a fallback topic that it subscribes to. Default is to use DVES_{last_part_of_macaddress}, I tried it and it works, so I can send commands to a single device.

I tried also without JSON but it seems to me it works in the same way, so using a shared topic to publish data (the real difference is that, without JSON, there are as many topics as values published, for example /VOLTAGE, /POWER and so on).

thomasf, 2016-12-12:
Each device should have it's own topic and you should usually name them in a
human readable way and then only use the topic name to identify which of your
sonoffs you are receiving from.

Naming your mqtt topics really helps in building the device map. As an example:
`sonoff/livingroom_light_ceiling2/power` is a lot easier to deal with than
having to translate the `DVES_XXXX` everywhere else, with a good naming scheme
you can even monitor your mqtt broker directly and understand what's going on
in your whole automation setup.

Another issue with sharing the telemetry (or any other) topic is that it's
impossible to retain messages on the mqtt broker if that's desired because only
the last sonoff which has posted would be retained.

The fallback name hints that it is not intended as a primary communication
channel, that should be your own configured topic.

thomasf, 2016-12-12:
...and you also have the group topic if you want to send messages to all your sonoffs.

jwildeboer, 2016-12-12:
IMHO the topology is irrelevant in the MQTT context. Devices should have a topic reflecting the device ID and the functions it offers. Nothing more. Mapping the device IDs to topology should be on a higher level.

With your approach you run into a problem when you have a device with multiple in/outputs that control things in different rooms. For example a device with 8 relays that controls 2 devices in the living room, 3 in the hallway and 3 in the kitchen. How are you going to name it?

IMHO the location/topology information is an attribute, not a topic.

guglielmino, 2016-12-12:
I agree, defining the topology on the device is also bad for scaling. Immagine when You have hundreds of devices, defining topology at upper level could be simpler than configuring the right topic in every device.

thomasf, 2016-12-12:
@jwildeboer I would name such a relay somehting like "central_switch_relay" if
it's located in my central automation rack... I would however prefer a solution
where a multi-relay is configurable with one topic per relay so that I can
decouple hardware from functionality ID.

I have no problem organizing 27 mqtt enabled devices here with functional role
naming, I just have to put in the device identity assignment in my provisioning
system rather than run time..

Regardless of this you should always assign a specific topic for each sonoff
device using this firmware. If you don't do that you will have the same problem
that you are experienceing now in all other aspects of your control system.. If
you dont want to use fuctional naming you should name then sonoff1, sonoff2,
sonoff3 or something like that instead. If you don't do this you will not be
able to turn on/off/toggle invidivual sonoff switches.

arendst, 2016-12-15:
There's been a lot of discussion here and I feel I have to elaborate on my topic subscribe decisions.

Initially I had one, by MQTT configurable topic planned called TOPIC but soon found out that two sonoffs come online with the same topic this would be a challenge at least...

I then introduced a unique, non-configurable topic which I call fallback topic that allows me to always  change the MQTT configurable topic to a new unique topic. This fallback topic is just what it is meant to be: A FALLBACK TOPIC in case of emergency.

All MQTT status messages will be send using the configurable TOPIC which should be made unique by the user. It might be called ```bedroom``` but it could also have been called ```titanic10``` as long as the user knows what it is and where to find it.

Having two sonoffs with the same topic allowed for MQTT commands to be send once to make them act in sono. That inspired me to add a third topic to subscribe to which I call GROUPTOPIC. Sonoffs with the same GROUPTOPIC will react to the same MQTT command. I use it to update firmware to all my sonoffs usings seperate groups for plain sonoff, pow sonoffs and th sonoffs.

So to come back to the initial question raised by Guglielmino I would suggest to change your default TOPIC to unique ones allowing you to recognize where the status information is coming from.

BTW this can be done online using the fallback topic and is only needed once. There is no need to change user_config.h all the time as many users seem to think they have to do with any new release. All changes are stored in flash and I make a lot of effort to keep these changes available between versions.

phedders, 2016-12-15:
> There's been a lot of discussion here and I feel I have to elaborate on my
> topic subscribe decisions.
>
> Initially I had one, by MQTT configurable topic planned called TOPIC but soon
> found out that [...]

Thanks for that summary - it all makes a lot of sense (as does your amazing firmware ;)

Please could you put this or a similar description in the main Readme? I spent a bit of time working out how this was best managed.

Thanks again.

arendst, 2016-12-17:
Explanation added to Tips on Wiki.

closed 2016-12-17 by arendst