arendst/Sonoff-MQTT-OTA-Arduino #139: Request ART-NET or OSC receive support

phedders, 2016-12-13:
I was wondering if ART-NET (or OSC) support could be added - I might have a look at doing it myself. It would allow Sonoffs to become part of a DMX/artnet universe. I would love this as an option instead of say the Domotix code if space were an issue.

[https://vvvv.org/contribution/artnet-arduino-set-v3.1-0](url) could be used - there would need to be persistent configuration of universe + channel (0-512).

thomasf, 2016-12-13:
For me, mqtt feels like one of the more convenient ways to integrate use as an integration point..  I think I have seen several osc/mqtt bridge softwares, sure it adds a tiny bit of latency but the wifi latency probably trumps all other latencies added anyways..

arendst, 2016-12-15:
I currently do not plan to integrate OSC in the software.

closed 2016-12-17 by arendst