arendst/Sonoff-MQTT-OTA-Arduino #140: Default UDP port and commands to turn the Sonoff Switch ON and OFF

WouterPieper, 2016-12-13:
I've long been searching and trying, but it still does not work, maybe someone can help me
I have a Sonoff Switch which I want to put on and off with UDP command.

I read a lot about the firmware update, but if that's not necessary, I would not do that. because I have not enough knowledge in this area

I use a loxone miniserver to send UDP commands.
I need to know which UDP port is used.

I'm also need the ASCII udp commands that i can use to turn the sonoff switch ON and OFF.

I can test the commands with a udp sender on my windows computer. 
For example i have another webrelais that already work. 
I attached a picture of this example.
To turn OFF this webrelais i use port 1200 and ASCII command: u=2password\n
To turn ON this webrelais i use port 1200 and ASCII command: a=2password\n

is there anyone who can tell me which port and commands for ON and OFF i can use for the sonoff switch?
Or is it not possible to turn the switch ON and OFF on this way without updated the firmware of the sonoff switch?

[image]

Lennyz1988, 2016-12-14:
You should find help at the forum of Itead. This has nothing to do with the firmware Sonoff-MQTT-OTA-Arduino by Arendst.

arendst, 2016-12-15:
As said by Lennyz this software does not support UDP other than the initial Amazon Alexa communication. 

For the original sonoff software you'll have to contact iTead.

closed 2016-12-15 by arendst