arendst/Sonoff-MQTT-OTA-Arduino #141: Send external switch status

rikki78, 2016-12-13:
With the USE_WALL_SWITCH configuration it is possible to use an external switch to GPIO14. Would it be possible to send the actual status of the switch as well, instead of only the relay status?

stoehrmark, 2016-12-17:
Also could this be coded to not only control the USE_WALL_SWITCH Command but just send a 1/0, true/false, etc. and not control the relay. Use it to just scene the switch state. For me it could be used with a reed switch indicating is a garage door is open or closed.

arendst, 2016-12-17:
Sending only the switch status is already available if you define a ButtonTopic different from your device topic. In that case it will send cmnd/<buttontopic>/POWER ON or OFF depending how you set SwitchMode.

The message ON or OFF is defined in user_config.h and can easily bechanged to 1 or 0.

rikki78, 2017-01-02:
Do you have any idea when this request will be implemented? 
Or perhaps I can implement it myself. Could you indicate what would be the best place for it? 
Would that be on the same place as send_button_power(1, flag); is now executed?

arendst, 2017-01-08:
See #206

closed 2017-01-08 by arendst

rikki78, 2017-01-08:
Hi,
I'm sorry to bother you again. My ideal situation would be as follows:
I would like to connect an external switch to the sonoff, when it toggles the relay toggles. But, this is done locally so in case of a controller failure the relay still toggles. On the other hand the switch status is still send so the controller so that can do something with the info (eg switch off after x minutes).
Is that possible? How do I need to set this up?
Thank you for helping!