arendst/Sonoff-MQTT-OTA-Arduino #142: After upgrade to 3.0.4 many MQTT commands are lost!

drhelmer, 2016-12-14:
I have been very happy with this software, and last revision installed was 2.0.11 I think, and I just upgraded to 3.0.4 today. Then suddenly I discovered that now and then the commands sent by my Openhab install or directly from an MQTT client doesnt reach the Sonoff! Now I have no clue as to why at this stage, but it is a bit worrying. Of course the usefulness of this stuff is dependent on all on/off switches to actually occur when sending the commands to the devices topic!

Any idea what is happening here?

arendst, 2016-12-15:
Please provide a list of exact MQTT topics that currently seem to fail you.

arendst, 2016-12-15:
I suppose you read the changes in command structure introduced in version 3.x and kind of documented in the _releasenotes.ino.

What you could do is disable Alexa support (// define USE_WEMO_EMULATION) in user_config.h, recompile and upgrade.

I'm currently doing some tests where I see MQTT reconnects which I think could be caused by the Wemo emulation using UDP which in turn receives a lot of messages.

drhelmer, 2016-12-15:
Hmm, actually strange you mention that, yes I noticed that and recompiled without define USE_WEMO_EMULATION, as I thought the reconnects could be an issue, but I am almost sure the lost MQTT commands where still lost. I ll have another try this afternoon. I enabled debug logging in console and noticed that the UDP messages were gone, so I didnt bother resetting config, should I have done that since I edited user_config.h or will such changes be picked up by a recompile?

drhelmer, 2016-12-15:
BTW the commands I noticed got lost where just "cmnd/<mytopic>/power on" and "off".

I had changed the topic through the webserver to "kjellerstueovn" in this particular case.

arendst, 2016-12-15:
Debug logging will remain as set as it is saved in flash and not overwritten by recompile.

drhelmer, 2016-12-15:
I wasn't referring to the logging, but the commenting of // define USE_WEMO_EMULATION. this is how the console log looks when I have commented that define \:
21:39:47 MQTT: Connected
21:39:48 MQTT: Attempting connection...
21:39:49 MQTT: Connected
21:39:49 MQTT: Attempting connection...
21:39:50 MQTT: Connected
21:39:51 MQTT: Attempting connection...

Before I commented I also got messages like UPnP: Multicast (re)joined

I tried sending MQTT commands this afternoon as well, the cmnd coommands are logged in Mosquitto, but no corresponding stat message is received from sonoff, and sonoff doesnt switch the relay..

drhelmer, 2016-12-15:
Here you see the log from the mosquitto_sub -v-t "#" \:

cmnd/sonoffs/webserver (null)
stat/kjellerstueovn/WEBSERVER Active for ADMIN on kjellerstueovn-5369 with IP address 192.168.0.15
cmnd/sonoffs/webserver (null)
stat/kjellerstueovn/WEBSERVER Active for ADMIN on kjellerstueovn-5369 with IP address 192.168.0.15
cmnd/sonoffs/webserver (null)
stat/kjellerstueovn/WEBSERVER Active for ADMIN on kjellerstueovn-5369 with IP address 192.168.0.15
cmnd/sonoffs/webserver (null)
stat/nisse/WEBSERVER Active for ADMIN on nisse-2159 with IP address 192.168.0.12
cmnd/sonoffs/webserver (null)
stat/nisse/WEBSERVER Active for ADMIN on nisse-2159 with IP address 192.168.0.12
cmnd/sonoffs/webserver (null)
stat/kjellerstueovn/WEBSERVER Active for ADMIN on kjellerstueovn-5369 with IP address 192.168.0.15
cmnd/sonoffs/webserver (null)
stat/nisse/WEBSERVER Active for ADMIN on nisse-2159 with IP address 192.168.0.12
cmnd/sonoffs/webserver (null)
stat/nisse/WEBSERVER Active for ADMIN on nisse-2159 with IP address 192.168.0.12
cmnd/sonoffs/webserver (null)
stat/nisse/WEBSERVER Active for ADMIN on nisse-2159 with IP address 192.168.0.12

The command I send to grouptopic sonoffs is \:

mosquitto_pub -t cmnd/sonoffs/webserver -m ""

It seems random which sonoff get the mqtt message and responds?

drhelmer, 2016-12-15:
I just tried the same message some 20 times \:
mosquitto_pub -t cmnd/sonoffs/webserver -m ""

The consistencyis that one of my 2 sonoffs responds to the groupmessage, but only one, and it seems random which one it is.

arendst, 2016-12-15:
First: seeing UPnP: Multicast (re)joined is normal as I disable UDP before every MQTT reconnect attempt.
Second: If I perform your command on 8 sonoffs I receive this:
```
cmnd/sonoffs/webserver (null)
stat/wkaku12/RESULT {"Webserver":"Active for ADMIN on wkaku12 with IP address 192.168.2.229"}
stat/wkaku10/RESULT {"Webserver":"Active for ADMIN on wkaku10 with IP address 192.168.2.226"}
stat/wkaku8/RESULT {"Webserver":"Active for ADMIN on wkaku8 with IP address 192.168.2.211"}
stat/wkaku11/RESULT {"Webserver":"Active for ADMIN on wkaku11 with IP address 192.168.2.223"}
stat/dual/RESULT {"Webserver":"Active for ADMIN on dual with IP address 192.168.2.228"}
stat/wkaku9/RESULT {"Webserver":"Active for ADMIN on wkaku9 with IP address 192.168.2.215"}
stat/sonoff3/RESULT {"Webserver":"Active for ADMIN on sonoff3 with IP address 192.168.2.202"}
stat/sonoff1/RESULT {"Webserver":"Active for ADMIN on sonoff1 with IP address 192.168.2.213"}
``` 
I think you have a MQTT server problem related to duplicate Client Ids. What may be your case is that your both sonoffs have the same CLIENT_ID and mosquito only handles one of them.

Are both sonoffs logging the same reception of your MQTT cmnd request?

drhelmer, 2016-12-15:
You were right of course. I had changed the MQTT_CLIENT_ID in user_config.h from 
#define MQTT_CLIENT_ID         "DVES_%06X"
to
#define MQTT_CLIENT_ID         "%06X"
in the status message  above from you I saw that the %06X was expanded with the last 6 characters in the hex MAC address, but in my status message it looked like below, so something didnt work in that change, although it should have worked, I dont see why at this moment. Anyhow, thanks a lot for all the help.

stat/nisse/RESULT MQT: Host 192.168.0.4, Port 1883, Client %06X (%06X), User DVES_USER, Password DVES_PASS, MAX_PACKET_SIZE 512, KEEPALIVE 15

One thing I dont get though, is that back in v. 2.0.15, you have code to check if the sysCfg.mqtt_client contains a %, and if so yuo copy the MAC address to it, I dont see that code anymore in 3.0.4, so I wonder how you initialize the variable if it contains the default "DVES_%06X" ?

Anyhow, thanks I can get it to work now :)

arendst, 2016-12-16:
%06X as MQTT_CLIENT_ID should have worked but it didn't - a bug that I've solved and will be in the next release.

closed 2016-12-17 by arendst