arendst/Sonoff-MQTT-OTA-Arduino #143: Deactivate LEDs blinkings in Normal-Mode

thommycgn, 2016-12-15:
The LEDs of the Sonoffs are really light and blinking from time to time. Not really nice in the sleeping room by night. It would be great if you could realize an option in Configuration Panel to prevent this.
Is it possible?

thomasf, 2016-12-15:
Maybe related to #136

0n3-70uch, 2016-12-15:
Hello, please Test this MQTT command:
LedState off

Look here for more informations:
https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/wiki/Commands

I don't know if this command will hot of the blinking but it is you can Test it.

Greetings Fabi

arendst, 2016-12-15:
Guys, been away for some days and this is on top of my list. ledstate off won't do it right now but I'll change it's options to allow disabling all blinks.

arendst, 2016-12-15:
Several options added to command LedState allowing more control over it's use in version 3.0.5 just released.

See wiki or _releasenotes.ino

closed 2016-12-17 by arendst