arendst/Sonoff-MQTT-OTA-Arduino #144: Error Version 3.0.5

Advant2, 2016-12-16:
00:00:15 Wifi: Connect failed with AP timeout
00:00:15 Wifi: **Connecting to AP255  () in mode 11N as sonoff-7719...**
00:00:30 Wifi: Connect failed with AP timeout
00:00:30 WPSconfig: Active for 1 minute
..after the upgrade, connects to WiFi obscure points, but the settings are registered point. Tell me what's going on?

arendst, 2016-12-16:
I guess you upgraded from a previous version. Please let me know what version.

For now perform a ```reset 1``` and try again.

Advant2, 2016-12-16:
I had a version 2_19a, I ask 3_05 and the switch is unable to connect to wifi. After that, I returned 2_19a firmware and it worked.

arendst, 2016-12-16:
Starting with version 3 another section of flash is being used containing your settings. After first start of 3.0.5 the flash settings from version 2.0.19a are being copied to the new area and updates introduced in version 3.x are amended using user_config.h

In your case it may be solved by first upgrading to version 2.1.2 to be found under releases and then upgrade to 3.0.5.

If that still doesn't work you'll have to update your version 3.0.5 user_config.h with your own parameters and change define CFG_HOLDER to something different from what it currently has. That way the new config will be used and all should be fine.

Advant2, 2016-12-16:
can you tell what a utility to completely clean esp8266 memory (512K), and once again write the firmware. Through the web page markups refuses to be updated. (Upload fayled  File upload buffer miskompare)

Advant2, 2016-12-16:
Program version	2.0.19a
Core/SDK version	2_3_0/1.5.3(aec24ac9)
Uptime	2 Hours
Flash write count	469
Boot count	105
Reset reason	Power on
 
SSId (RSSI)	Home_NET (38%)
Hostname	sonoff-7719
IP address	192.168.1.124
Gateway	192.168.1.1
MAC address	18:FE:34:9F:BE:27
 
MQTT Host	192.168.1.100
MQTT Port	1883
MQTT Client and
 Fallback Topic	DVES_9FBE27
MQTT User	DVES_USER
MQTT Topic	sonoff
MQTT Group Topic	sonoffs
 
ESP Chip id	10468903
Flash Chip id	1261768
Flash size	512kB
Program flash size	512kB
Program size	379kB
Free program space	48kB
Free memory	29kB

Advant2, 2016-12-16:
and one more question: how to set a static IP address?

arendst, 2016-12-16:
First: Only DHCP IP addresses are supported.
Second: Your hardware does nor support my software as it needs at least 1MB Flash and your hardware only has 512kB flash. This does not seem to be iTead sonoff hardware.

Advant2, 2016-12-16:
Ok, I try to understand the other device. Thank you very much for your help.

closed 2016-12-17 by arendst