arendst/Sonoff-MQTT-OTA-Arduino #146: Version 3.0.7 not connecting to WIFI

connectedthings, 2016-12-18:
Must have missed something ... have been working with version 2.x for a while but when I convert an existing module to 3.x the module does not connect to wifi. All settings in userconf.h same as for version 2.x. The only other change I made was setting wifimanager as default config fallback.

When I connect via wifimanager after first boot (when modules fails to connect) the second AP entry is garbage. Complete all fields correctly and setting are retained on reboot but wifi never connects.

This happens for both new modules (flash from IDE) and OTA from 2.x to 3 on existing modules. 

If I flash back to 2.x from IDE all ok again ....

Any ideas?

hhcab79, 2016-12-18:
I have exactly the same issue, with the same symptoms as connected things. Would really appreciate any pointers / advice. Many thanks for your great work.

alphalove, 2016-12-18:
Hello,
Try resetting the stored config settings by holding the button down for 4 seconds (restores the default / compiled user_config), then run through the whole setup from scratch.  I thought I had some odd behaviour going from 2.x to 3.x but everything is fine now.

hhcab79, 2016-12-18:
Hi alphalove,
That worked a treat. Thanks very much for the advice.

arendst, 2016-12-18:
The upgrade from 2.x to 3.x was supposed to go smoothly as long as you wouldn't change your 3.x user_config.h...

3.x uses another area of flash and rearranges the settings too. It migrates all settings from 2.x and adds the ones from 3.x which in your case would be wrong but you should be able to come online again. When online you should change settings using commands either via MQTT, webconsole or serial.

If this fails try indeed to hold the button for more than 4 seconds to do a reset of parameters using the user_config.h values.

If that fails too it is now possible (as the initial values have been migrated) to update your 3.x user_config.h with your desired values AND CHANGE DEFINE CFG_HOLDER too. When this is loaded you're ready for future 3.x updates.

Sorry it became a mess. I've tried to overcome this by auto migrating settings from 2.x to 3.x using all  scenarios I could think off but as always there is more to life...

BTW your 2.x setting remain intact so you can always go back. Auto migrating won't work anymore as the 3.x area has now been initialised with possible rubish. The only way to fix that is using the changed CFG_HOLDER define as mentionend above.

connectedthings, 2016-12-18:
No mess at all Theo, no one would expect you to cover all scenarios on your lonesome! Now see how it works and will play with the CFG define. The 4s reset did the trick to get things going again.

Tx again for the great work and superb support.

closed 2016-12-18 by connectedthings