arendst/Sonoff-MQTT-OTA-Arduino #147: IP address, overview?

timtimsson, 2016-12-18:
Hello, 
i use a lot of devices. how can i find out the different ip (we use dhcp)? Is it possible to add a entry at tele/stat or can you implement a command that each device answer with his own address?
Thanks for your fantastic work!

arendst, 2016-12-18:
If your grouptopic is set to ```sonoffs``` you can use ```cmnd/sonoffs/status 5``` to dump all network information of all sonoffs.
```
cmnd/sonoffs/status 5
stat/wkaku12/RESULT {"StatusNET":{"Host":"wkaku12", "IP":"192.168.2.229", "Gateway":"192.168.2.254", "Subnetmask":"255.255.255.0", "Mac":"5C:CF:7F:1A:96:43", "Webserver":2, "WifiConfig":4}}
stat/wkaku10/RESULT {"StatusNET":{"Host":"wkaku10", "IP":"192.168.2.226", "Gateway":"192.168.2.254", "Subnetmask":"255.255.255.0", "Mac":"5C:CF:7F:88:83:B0", "Webserver":2, "WifiConfig":0}}
stat/wkaku11/RESULT {"StatusNET":{"Host":"wkaku11", "IP":"192.168.2.223", "Gateway":"192.168.2.254", "Subnetmask":"255.255.255.0", "Mac":"5C:CF:7F:85:E4:88", "Webserver":2, "WifiConfig":4}}
stat/wkaku8/RESULT {"StatusNET":{"Host":"wkaku8", "IP":"192.168.2.211", "Gateway":"192.168.2.254", "Subnetmask":"255.255.255.0", "Mac":"5C:CF:7F:88:75:AD", "Webserver":2, "WifiConfig":4}}
stat/wkaku9/RESULT {"StatusNET":{"Host":"wkaku9", "IP":"192.168.2.215", "Gateway":"192.168.2.254", "Subnetmask":"255.255.255.0", "Mac":"5C:CF:7F:1C:0F:46", "Webserver":2, "WifiConfig":4}}
stat/wemos5/RESULT {"StatusNET":{"Host":"wemos5", "IP":"192.168.2.230", "Gateway":"192.168.2.254", "Subnetmask":"255.255.255.0", "Mac":"5C:CF:7F:13:9F:3D", "Webserver":2, "WifiConfig":3}}
stat/sonoff1/RESULT {"StatusNET":{"Host":"sonoff1", "IP":"192.168.2.213", "Gateway":"192.168.2.254", "Subnetmask":"255.255.255.0", "Mac":"5C:CF:7F:0C:A5:5C", "Webserver":2, "WifiConfig":4}}
stat/sonoff3/RESULT {"StatusNET":{"Host":"sonoff3", "IP":"192.168.2.202", "Gateway":"192.168.2.254", "Subnetmask":"255.255.255.0", "Mac":"5C:CF:7F:0C:AB:6F", "Webserver":2, "WifiConfig":4}}
stat/dual/RESULT {"StatusNET":{"Host":"dual", "IP":"192.168.2.228", "Gateway":"192.168.2.254", "Subnetmask":"255.255.255.0", "Mac":"60:01:94:07:24:E3", "Webserver":2, "WifiConfig":4}}
```

timtimsson, 2016-12-18:
I've overlooked - Thanks

closed 2016-12-18 by timtimsson