arendst/Sonoff-MQTT-OTA-Arduino #148: Having trouble changing WIFI id

timt1961, 2016-12-18:
Total Noob at flashing here.. 
I'm trying to flash a sonoff module with the firmware.
First attempt was to flash with unmodified code. This went well, and the serial console shows the unmodified access points. 

Changed the STA_SSID1 and STA_PASS1. connnected the sonoff, while holding down the button and started the flash.

However, I still see this:
0:00:00 APP: Project sonoff (Topic sonoff, Fallback DVES_07A50A, GroupTopic sonoffs) Version 3.0.8
00:00:00 Wifi: Connecting to AP2 indebuurt2 (VnsqrtnrsddbrN) in mode 11N as sonoff-1290...
00:00:07 Wifi: Connect failed as AP cannot be reached
00:00:07 Wifi: Connecting to AP1 indebuurt1 (VnsqrtnrsddbrN) in mode 11N as sonoff-1290...

Note the version number.. I've changed that to 3.0.**8**, to check that my new version is being uploaded.. 

So.. what am I doing wrong ? Is there something like a "make clean" that I'm missing ?

arendst, 2016-12-18:
Changes to user_config.h have no effect unless you ALSO CHANGE CFG_HOLDER. Without this change flash parameters are being used.

In your case why don't you just execute commands ssid1 and password1 with your access point parameters for a quick test

timt1961, 2016-12-18:
Like I said: total Noob..missed that bit.
But if I read the CFG_HOLDER correctly, I'd just change this to a local value ? (and bump that everytime I change something?)
And otherwise, A four-second press will reset to the values in user_config.h.. I just found the relevant page.
Should be able to get on track now.

Thanks for the help

arendst, 2016-12-18:
CFG_HOLDER only needs to be changed for settings you want to change in flash which should almost never happen as you can change them using their specific commands.

Changing options like MODULE or enabling sensors do not need a change to CFG_HOLDER.

closed 2016-12-21 by arendst