arendst/Sonoff-MQTT-OTA-Arduino #149: compile

zschramm, 2016-12-18:
Sorry if this is a dumb question, but I cloned the folders and updated user_config.h and when I compile with settings as described, I get:

`'addLog_P'` was not declared in this scope``

Typically this means a missing semi-colon or brace not closed, or library reference?  I downloaded the code directly so I would assume it was working?   These little things always seem to get me. 

I also get all these other errors during compile:
D:\Dropbox\Documents\Arduino\Sonoff-MQTT-OTA-Arduino-master\sonoff\sonoff.ino: In function 'void CFG_Default()':

sonoff:405: error: 'addLog_P' was not declared in this scope

   addLog_P(LOG_LEVEL_NONE, PSTR("Config: Use default configuration"));

                                                                     ^

sonoff:424: error: 'STA_SSID2' was not declared in this scope

   strlcpy(sysCfg.sta_ssid[1], STA_SSID2, sizeof(sysCfg.sta_ssid[1]));

                               ^

sonoff:425: error: 'STA_PASS2' was not declared in this scope

   strlcpy(sysCfg.sta_pwd[1], STA_PASS2, sizeof(sysCfg.sta_pwd[1]));

                              ^

sonoff:488: error: 'CFG_Save' was not declared in this scope

   CFG_Save();

            ^

D:\Dropbox\Documents\Arduino\Sonoff-MQTT-OTA-Arduino-master\sonoff\sonoff.ino: In function 'void CFG_Migrate_Part2()':

sonoff:493: error: 'addLog_P' was not declared in this scope

   addLog_P(LOG_LEVEL_NONE, PSTR("Config: Migrating configuration"));

                                                                   ^

sonoff:577: error: 'CFG_Save' was not declared in this scope

   CFG_Save();

            ^

D:\Dropbox\Documents\Arduino\Sonoff-MQTT-OTA-Arduino-master\sonoff\sonoff.ino: In function 'void mqtt_publish(const char*, const char*, boolean)':

sonoff:723: error: 'addLog' was not declared in this scope

   addLog(LOG_LEVEL_INFO, log);

                             ^

D:\Dropbox\Documents\Arduino\Sonoff-MQTT-OTA-Arduino-master\sonoff\sonoff.ino: In function 'void mqtt_connected()':

sonoff:811: error: 'spiffsPresent' was not declared in this scope

     if (!spiffsPresent()) {

                        ^

D:\Dropbox\Documents\Arduino\Sonoff-MQTT-OTA-Arduino-master\sonoff\sonoff.ino: In function 'void mqtt_reconnect()':

sonoff:851: error: 'mqttDataCb' was not declared in this scope

     mqttClient.setCallback(mqttDataCb);

                            ^
[... 607 lines cut ...]
D:\Dropbox\Documents\Arduino\Sonoff-MQTT-OTA-Arduino-master\sonoff\webserver.ino: In function 'void handleNotFound()':

webserver:1159: error: 'captivePortal' was not declared in this scope

   if (captivePortal()) { // If captive portal redirect instead of displaying the error page.

                     ^

D:\Dropbox\Documents\Arduino\Sonoff-MQTT-OTA-Arduino-master\sonoff\webserver.ino: In function 'boolean captivePortal()':

webserver:1183: error: 'isIp' was not declared in this scope

   if (!isIp(webServer->hostHeader())) {

                                    ^

Using library Ticker at version 1.0 in folder: C:\Users\Zac\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\Ticker 
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Zac\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi 
Using library ESP8266HTTPClient at version 1.0 in folder: C:\Users\Zac\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266HTTPClient 
Using library ESP8266httpUpdate at version 1.0 in folder: C:\Users\Zac\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266httpUpdate 
Using library arduino_77356 at version 2.4 in folder: D:\Dropbox\Documents\Arduino\libraries\arduino_77356 
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\Zac\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer 
Using library DNSServer at version 1.1.0 in folder: C:\Users\Zac\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\DNSServer 
exit status 1
'addLog_P' was not declared in this scope

arendst, 2016-12-18:
You might take a look in wiki at the top of this page...

zschramm, 2016-12-18:
Thank you for that informative reply.  I did go through the wiki and I did not see anything addressing this issue.  Can you elaborate?

arendst, 2016-12-18:
Suppose the code on github is correct. I know it's a bold statement but suppose it is true.

After reading the wiki, installing the correct environment and downloading the correct package (you might consider the releases tab) and you still receive these error messages then you might try a simple blink routine to your sonoff first.

arendst, 2016-12-18:
Further on the previous answer I notice you are using version 2.0.0 of the esp8266 arduino code.

As you stated you read the wiki I kindly request to do it again this starting from the prerequisite page where it is stated that AT LEAST VERSION 2.3.0 is needed.

Happy upgrading.

arendst, 2016-12-18:
Hope it's informative enough. RTFW.

zschramm, 2016-12-20:
Yes that information was informative and I was able to flash the firmware.  I would have never figured it out in a million years that I needed to update the esp8266 library, the arduino IDE itself (it works on 1.6.18 btw).  

So thank you for your time and help on this.  I could have done without the sarcasm and just your 3rd response which solved it for me.

closed 2016-12-20 by arendst

tringler, 2017-02-19:
I have exactly the same compiler errors using Arduino 1.8.1 and esp8266 library 2.3.0 - I fulfilled all other requirement like putting in IRremoteESP8266, PubSubClient and NeoPixelBus into my lib folder. I also installed ArduinoJson. I would say I RTFW. Any idea to get rid of these compiler errors?