arendst/Sonoff-MQTT-OTA-Arduino #150: add #error vor MQTT_PACKET_SIZE and TTL

hjgode, 2016-12-19:
Please add to sonoff.ino:

...
#include <PubSubClient.h>                   // MQTT
#if (MQTT_MAX_PACKET_SIZE < 400)
  #error "MQTT_MAX_PACKET_SIZE is less than 400. Edit PubSubClient.h"
#endif
#if (MQTT_SOCKET_TIMEOUT < 120)
  #error "MQTT_SOCKET_TIMEOUT is less than 120. Edit PubSubClient.h"
#endif
...

To avoid ending with a non-working firmware file

~Josef

arendst, 2016-12-19:
Josef,

do you really experience non-working firmware when these parameters are not set correctly? I did not encounter that problem and that's why the test and message is included in the first messages after reboot in both serial console and MQTT.

thomasf, 2016-12-19:
why not just set MQTT_MAX_PACKET_SIZE before including PubSubClient.h ?

arendst, 2016-12-19:
Thomas see #87 and others. Just doesn't work in Arduino IDE.

thomasf, 2016-12-19:
Hmm.. by looking at the cpp file generated by the Arduino preprocessor it might work if the define is added to the top of the main `sonoff.ino` file because it seems to be put at the top..

```cc
#include <Arduino.h>
#line 1 "/home/thomasf/src/github.com/arendst/Sonoff-MQTT-OTA-Arduino/sonoff/sonoff.ino"
#line 1 "/home/thomasf/src/github.com/arendst/Sonoff-MQTT-OTA-Arduino/sonoff/sonoff.ino"
/*
 * Sonoff and ElectroDragon by Theo Arends
 *
 * ====================================================
 * Prerequisites:
 *   - Change libraries/PubSubClient/src/PubSubClient.h
 *       #define MQTT_MAX_PACKET_SIZE 400
 *
 *   - Select IDE Tools - Flash size: "1M (64K SPIFFS)"
 * ====================================================
*/

#define VERSION                0x03000900   // 3.0.9

#define SONOFF                 1            // Sonoff, Sonoff SV, Sonoff Dual, Sonoff TH 10A/16A, S20 Smart Socket, 4 Channel
#define SONOFF_POW             9            // Sonoff Pow
#define ELECTRO_DRAGON         10           // Electro Dragon Wifi IoT Relay Board Based on ESP8266

#define DHT11                  11
#define DHT21                  21
#define DHT22                  22
#define AM2301                 21
#define AM2302                 22
#define AM2321                 22

enum log_t   {LOG_LEVEL_NONE, LOG_LEVEL_ERROR, LOG_LEVEL_INFO, LOG_LEVEL_DEBUG, LOG_LEVEL_DEBUG_MORE, LOG_LEVEL_ALL};
enum week_t  {Last, First, Second, Third, Fourth};
enum dow_t   {Sun=1, Mon, Tue, Wed, Thu, Fri, Sat};
enum month_t {Jan=1, Feb, Mar, Apr, May, Jun, Jul, Aug, Sep, Oct, Nov, Dec};
enum wifi_t  {WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, MAX_WIFI_OPTION};
enum msgf_t  {LEGACY, JSON, MAX_FORMAT};
enum swtch_t {TOGGLE, FOLLOW, FOLLOW_INV, PUSHBUTTON, PUSHBUTTON_INV, MAX_SWITCH_OPTION};
enum led_t   {LED_OFF, LED_POWER, LED_MQTTSUB, LED_POWER_MQTTSUB, LED_MQTTPUB, LED_POWER_MQTTPUB, LED_MQTT, LED_POWER_MQTT, MAX_LED_OPTION};

#include "user_config.h"
#include "user_config_override.h"

/*********************************************************************************************\
 * Enable feature by removing leading // or disable feature by adding leading //
\*********************************************************************************************/

//#define USE_SPIFFS                          // Switch persistent configuration from flash to spiffs (+24k code, +0.6k mem)

/*********************************************************************************************\
 * Not released yet
\*********************************************************************************************/

#if MODULE == SONOFF_POW
//  #define FEATURE_POWER_LIMIT
#endif
#define MAX_POWER_HOLD         10           // Time in SECONDS to allow max agreed power
#define MAX_POWER_WINDOW       30           // Time in SECONDS to disable allow max agreed power
#define SAFE_POWER_HOLD        10           // Time in SECONDS to allow max unit safe power
#define SAFE_POWER_WINDOW      30           // Time in MINUTES to disable allow max unit safe power
#define MAX_POWER_RETRY        5            // Retry count allowing agreed power limit overflow

/*********************************************************************************************\
 * No user configurable items below
\*********************************************************************************************/

#define SONOFF_DUAL            2            // (iTEAD PSB)
#define CHANNEL_3              3            // iTEAD PSB
#define CHANNEL_4              4            // iTEAD PSB
#define CHANNEL_5              5            // Future use
[... 51 lines cut ...]

#include "support.h"                        // Global support
#include <Ticker.h>                         // RTC
#include <ESP8266WiFi.h>                    // MQTT, Ota, WifiManager
#include <ESP8266HTTPClient.h>              // MQTT, Ota
#include <ESP8266httpUpdate.h>              // Ota
#include <PubSubClient.h>                   // MQTT
#ifdef USE_WEBSERVER
  #include <ESP8266WebServer.h>             // WifiManager, Webserver
  #include <DNSServer.h>                    // WifiManager
#endif  // USE_WEBSERVER
#ifdef USE_SPIFFS
  #include <FS.h>                           // Config
#endif
#ifdef SEND_TELEMETRY_I2C
  #include <Wire.h>                         // I2C support library
#endif // SEND_TELEMETRY_I2C

typedef void (*rtcCallback)();

extern "C" uint32_t _SPIFFS_start;
extern "C" uint32_t _SPIFFS_end;
...

```

arendst, 2016-12-19:
Well give it a try, test it thoroughly and if it works I'll implement but I won't spent any time on it anymore ;-)

hjgode, 2016-12-20:
> Josef,
> do you really experience non-working firmware when these parameters are not set correctly? I did not > encounter that problem and that's why the test and message is included in the first messages after
> reboot in both serial console and MQTT.

I thought this was the problem of 3.0.9 not running on my sonoffs. The firmware always crashed with an exception (3) and rebooted after wdt. One issue (self made? or firmware quirk?) was that my mosquitto server is set to retain and for whatever reason, cmnd/sonoff4/upgrade 1 was set with qos=1. That resulted in letting the sonoff4 (tested with other sonnoff too) do an upgrade from http, doing a reboot before the mqtt message had been acknowledged. That resulted in a boot/upgrade/crash loop. This stopped after I deleted the mosquitto.db file.

But this is not all, the 3.0.9 still does crash with exception(3) and wdt over and over. Finally I went back to 2.0.10 and the sonoffs run fine again know. Never touch a running system ;-)

thanks for all the code and patience.

For me it is OK if the MQTT_PACKET_SIZE warning reamins as is.

BTW:
a) do we really need the warning about MQTT_PACKET_SIZE etc?
b) putting another, same #define in the code before PubSubClient.h will result in error 'already defined'
c) putting the #define after pubsubclient.h makes it useless

~josef

thomasf, 2016-12-20:
Nope.. my pubsubclient.h seems to be correctly written: 

```cc
// MQTT_MAX_PACKET_SIZE : Maximum packet size
#ifndef MQTT_MAX_PACKET_SIZE
#define MQTT_MAX_PACKET_SIZE 128
#endif

// MQTT_KEEPALIVE : keepAlive interval in Seconds
#ifndef MQTT_KEEPALIVE
#define MQTT_KEEPALIVE 15
#endif

// MQTT_SOCKET_TIMEOUT: socket timeout interval in Seconds
#ifndef MQTT_SOCKET_TIMEOUT
#define MQTT_SOCKET_TIMEOUT 15
#endif
...
```

arendst, 2016-12-20:
Josef, the change in MQTT_PACKET_SIZE is not needed for normal power on/off operation but for messages longer than the default of 128 characters it becomes a problem as not part of the message is send but the whole message is skipped if it is longer than 128 characters. The status messages would be longer messages as are the commands messages.

About the exception 3, what happens if you press and hold the button for over 4 seconds (Should reset migrated flash to user_config.h and hopefully stops running in exception)?

Thomas, have you checked it's functionality? Are you able to receive messages longer than 128 characters if you add your redefine of MQTT_MAX_PACKET_SIZE in sonoff.ino? (Answer probably NO).

hjgode, 2016-12-20:
Hello Arend

during the MQTTdisaster with the retained Upgrade cmnd pressing and holding to load defaults did not help. Possibly as there was also a cmnd/sonoff/upgrade 1 in the air... The device restarted itself to fast for being accessed.

Now I am running only one sonoff with v3.0.9 and the rest (currenty 4 more) with 2.0.10. I assume 3.0.9 will run without issues except for the strange http connect (no host). I have to enter the ip several times in the browser and do Retry, before I get a connection. But this is nothing urgent.

Thanks for care

Josef

hjgode, 2016-12-21:
I just realized that if compiled with platformio, the MQTT_MAX_PACKET_SIZE is already changed to 400 in platformio.ini:

`build_flags = -Wl,-Tesp8266.flash.1m64.ld -DMQTT_MAX_PACKET_SIZE=400`

If one compiles with Arduino IDE, the default packet size is used unless the following line is inserted in sonoff.ino BEFORE #include 'PubSubClient.h":

`#define MQTT_MAX_PACKET_SIZE 400
`
The MQTT_MAX_PACKET_SIZE in PubSuClient.h and others were changed to be externally assignable after the discussion here: https://github.com/knolleary/pubsubclient/issues/110

I leave it up to you to include a compile #warning, #message or #error in sonoff.ino to at least inform the user before he sees the warning in the console messages (or not) of a running firmware.

~josef

closed 2016-12-21 by hjgode