arendst/Sonoff-MQTT-OTA-Arduino #151: Serial Commands

scargill, 2016-12-20:
Late at night and I've blown a Sonoff - can't remember the serial commands..

The unit is called "smart" - want to change it to smart1 so I tried:

cmnd/smart/topic smart2

IMMEDIATELY crashes the board ever time...

So my question is - what did I do wrong - and why would a simple error of serial input cause the board to crash?

arendst, 2016-12-21:
Pete,

changing topic using serial commands is done by typing ```topic smart2``` without the ```cmnd/smart/``` part. Using MQTT you would indeed use your commandline ```cmnd/smart/topic smart2```.

I've tried your commandline and it indeed performs NOT as I expected. I'll have to investigate but it looks like a buffer overflow is causing problems.

So for the moment it's both a user error (yours) and a design error (mine) ;-)

scargill, 2016-12-21:
Correct on both accounts - I was doing it wrong - now working... and I'm pretty sure you're right about buffer overflow... would be nice to tidy that up.. my software repies "eh?" in such circumstances..

arendst, 2016-12-21:
Solved in version 3.1.0 just released

scargill, 2016-12-21:
Well done.

closed 2016-12-21 by arendst