arendst/Sonoff-MQTT-OTA-Arduino #152: Console commands not working

tam481, 2016-12-21:
Hello,
Once again, thanks for the amazing firmware.

I haven't been able to send commands to Sonoff since version 3.0.3. I don't think it's that particular build. It's just when I noticed the issue.

I've tried resetting to default configs to no avail. Cleared cookies, tried a different browser but nothing happens when I enter commands and hit enter.
I've tried 
cmnd/home/bedroom/sonoff Reset 2
cmnd/home/bedroom/sonoff Reset2
Reset 2
Restart
cmnd/home/bedroom/sonoff Restart

In the past entering the wrong syntax would result in an MQTT error now nothing happens.

I've also tried overwriting the firmware with NodeMCU then loading this firmware again to no avail.

I'm now running 3.1.0 and still experiencing the same issue. The reason why I tried this is because MQTT keeps turning the light ON after upgrading regardless even if the "retained" message is OFF and I wanted to do a complete "Reset 2" to clear everything.

tam481, 2016-12-21:
Using MQTT-Spy
I can see that the status of cmnd and stat can have a different status so that the last cmnd is ON while stat is OFF which is odd given that stat should reflect the last message sent by the last cmnd!

However, the issue with the relay turning on automatically after an upgrade persists still even when cmnd and stat are both set to OFF so I'm not sure how MQTT could be causing this issue.

I hope this helps. 

I know the "issue" is to do with commands but I stumbled upon that while trying to fix the automatic ON issue.

tam481, 2017-01-06:
Hello all,
Update: managed to resolve the MQTT issue by deleting my MQTT db and starting again which is fine as I don't have that many devices.

However, I cannot get Console commands to work no matter what browser I use. I used to modify the user_config.h file but moved to using the override file in the hope of fixing whatever is broken to no avail. What can I do to get the Console to work?

I enter a command then hit enter and nothing happens.

arendst, 2017-01-06:
You might try to execute command ```weblog 2``` in the console...

tam481, 2017-01-06:
Tried that to no avail!
I can see the hourly tele status reports

tam481, 2017-01-06:
I can publish the command via MQTT and I can see it in the console after publishing it via MQTT. It's jus the web version that doesn't work.

arendst, 2017-01-06:
What is the exact text you type in the console?

tam481, 2017-01-06:
weblog 2

Also tried 

cmnd/home/livingroom/TableLamp/weblog 2

arendst, 2017-01-06:
First stop using the complete topic as there is no need for it. It already knows it is on cmnd/home/livingroom/TableLamp.

Second how did you configure the cmnd/home/livingroom/TableLamp?
What part is defined in SUB_PREFIX and what part in TOPIC?

tam481, 2017-01-06:
I had been using the "shortened" but thought I'd try the full thing.

I'm using the following in the override file:
#ifdef PROJECT
#undef PROJECT
#endif
#define PROJECT                "TableLamp"

#ifdef SUB_PREFIX
#undef SUB_PREFIX
#endif
#define SUB_PREFIX             "cmnd/home/livingroom"

arendst, 2017-01-06:
Do you see more information if you execute ```weblog 3```?

tam481, 2017-01-06:
No. It's as if nothing happened.

tam481, 2017-01-06:
Tried with my second Sonoff and getting the same thing i.e. nothing happens

arendst, 2017-01-06:
And if you execute it from your mqtt client: cmnd/home/livingroom/TableLamp/weblog 3 (the topic is case sensitive!)

tam481, 2017-01-06:
I get this when I do it via the mosquitto_pub utility. 

16:58:08 MQTT: stat/home/livingroom/TableLamp/WEBLOG = 3
16:58:08 Config: Saved configuration (928 bytes) to flash at F8 and count 39

arendst, 2017-01-06:
That's correct. Now try from the console command ```status```. What do you see in the console and what does the MQTT server see?

tam481, 2017-01-06:
now `status` returns:
17:07:41 HTTP: Handle console

tam481, 2017-01-06:
In fact, any command such as weblog 3 and weblog 2 returns Handle console

arendst, 2017-01-06:
That's correct as it is the extra information enabled by option 3 of the weblog command. Puzzling is why the other messages are not being shown...
```
17:09:22 HTTP: Handle console
17:09:22 MQTT: DataCb Topic sonoff, Group 0, Index 1, Type STATUS, Data  ()
17:09:22 MQTT: stat/sonoff/RESULT = 3.1.11b, 1, sonoff, 0, POWER, 0, 3, 1, 1, 1, 1, 0, 0
```

BuilderBrian, 2017-01-06:
I am brand new here as of this morning but when I was setting things up
this is what I found….

The back slashes in the topic created an issue for me.

I was using “Home/LivingRoom/Couch” and nothing worked.

With a simple topic of “Sonoff” everything work.

I then used a topic of “Home_LivingRoom_Couch” and everything started
working again.

I am not sure if this is related to your issue or not….

I am only a handful of hours into my use of this software so I may be way
off base J

Brian

arendst, 2017-01-06:
I can reproduce! If I use your PUB_PREFIX (cmnd/home/livingroom) and SUB_PREFIX (stat/home/livingroom) it fails too.

Current workaround: do not change the default PUB_PREFIX and SUB_PREFIX or at least do not use ```/``` in it.

tam481, 2017-01-06:
That's great. Thank you @arendst and also @BuilderBrian. I'll change it to use the default until you fix it in a future release :-)

arendst, 2017-01-07:
Fixed in version 3.1.13 just released

closed 2017-01-08 by arendst