arendst/Sonoff-MQTT-OTA-Arduino #155: Support for Itead Sonoff SC

ThomDietrich, 2016-12-23:
Hey Theo,
back with yet another new product.

* https://www.itead.cc/sonoff-sc.html
* https://www.itead.cc/wiki/Sonoff_SC

Are you personally interested in extending your firmware for it?

arendst, 2016-12-23:
No not at the moment but that may change...

ThomDietrich, 2016-12-23:
I'd actually be willing to motivate this issue by sponsoring one device for you (or one half).
Is this something you are interested in or would you rather not receive donations for this open source project?
Best! Thomas

jaysuk, 2016-12-24:
I just want to make sure you know this isn't a typical sonoff product. It doesn't run the sensors off an ESP. It has an atmega328 chip connected to the sensors which then feeds the data to an ESP to transmit the data to the internet. 
The code for the atmega328 is available on iteads wiki but there is no code for the ESP. 
It the ESP would have to be programmed to collect the data a d transmit it that way, assuming it's useable data. The temperature sensor is a rubbish DHT11

ThomDietrich, 2016-12-24:
I've looked at the data sheet and the schematic. I don't think it will be remarkably hard to adopt arendst firmware for that. Of course we can only be sure after it was tested.
The DHT11 is sure not the best sensor but I wouldn't define it as useless or rubbish...

arendst, 2016-12-24:
I've read somewhere that they are already planning to replace the DHT11 by a DHT22. Still no high-tech but at least better than before.

I might receive a unit so hold on with your funding...

ThomDietrich, 2016-12-24:
The only thing making me sad is, that the unit doesn't have a CO2 sensor. I'd expect the "environment monitor" in my bedroom to be equipped with one :-/ But it's no wonder, $19.90 is not enought to even buy one of these on the street.

derlucas, 2016-12-27:
i ordered two of that Sonoff SC units and can supply hires photos of this.

farconada, 2016-12-28:
I'm very interested in it too
It will be great to installarends  Sonoff firmware

derlucas, 2016-12-28:
I started a firmware for Sonoff SC based on Homie-ESP8266. Maybe you can start from there:
https://github.com/derlucas/homie-sonoff/tree/master/sonoff-homie-sc

I changed the original ATMega328p firmware to tell the sensor values on a regular base. The ESP runs Homie Firmware and sends the Data from Serial over to MQTT.

farconada, 2016-12-28:
IMHO the esp8266 firmware should respect the original interface with the arduino firmware provided by itead, so you could have separate developments

farconada, 2017-01-04:
I've placed my bounty to buy a Sonoff SC to @arendst

xoseperez, 2017-01-12:
I have also been working on the Sonoff SC, with a firmware based on my ESPurna firmware for Sonoff devices and alike.

I have added DHT22 support, MQTT and Domoticz integration. Also I'm using the mic to monitor claps and send custom MQTT messages so you can clap your lights on/off. 

An explanation of the whole process is here:  http://tinkerman.cat/sonoff-sc-with-mqtt-and-domoticz-support/

JoeWiseman, 2017-01-29:
I know it might not be the right place to ask, but is there any way to use a brighness sensor along with the existing Sonoff-devices (e.g. Sonoff TH) in combination with the firmware given here?
I've red that the "brightness sensor" just give the a "dusky, normal and bright", so it would be better to get a numerical value in e.g. lux.
Would be great if anyone did some experience with that and is willing to share it.
Regards
Joe

ThomDietrich, 2017-04-10:
First support via: https://github.com/arendst/Sonoff-Tasmota/commit/79b63858b9c67f8bd33b8b5ce417702a294f6bc8, see https://github.com/arendst/Sonoff-Tasmota/issues/112

closed 2017-04-10 by ThomDietrich