arendst/Sonoff-MQTT-OTA-Arduino #157: "File upload buffer miscompare" when OTA a modified firmware

Nayar, 2016-12-26:
Hi,

In what conditions do I get "File upload buffer miscompare" when uploading a new firmware? I added a functions and it didn't want to upload. When flashing to device through wire, it seems to work.

arendst, 2016-12-26:
You compiled with wrong flash option; either 512 flash or no or  to much spiffs.

eifinger, 2016-12-27:
Hi there, I just wanted to upload version 3.1.3 via File Upload and got the same error. I made sure I compiled the file with the correct flash options.
Will now try the other OTA method.

eifinger, 2016-12-27:
Thank you very much for clarifying that.
Works like a treat now.

Thank you for the support and the awesome software!

closed 2016-12-27 by arendst

julred, 2018-05-19:
Hi,
I have the same issue.
I flashed the latest spftware 5.14.0 via Arduino with the following preferences:
Flash mode: DOUT
Flash size: 1M (64k SPIFFS)

This is extracted from the information page:

ESP Chip Id        11028069
Flash Chip Id     1327198
Flash Size          1024kB
Program Flash Size  1024kB
Program Size  560kB
Free Program Space  376kB
Free Memory 17kB

And trying to upload a binary file (sonoff-DE.bin) generates the following error message:
Upload Failed

Upload buffer miscompare

littlebilly, 2018-05-21:
Hi,
I have the same issue with 5.14.0
Flashing Version 5.12.0i-2_3_0 works without any error message.
This thread is closed? Maybe you should open a new one?
Billy

microspino, 2018-06-02:
@littlebilly you saved my day.
Uploading sonoff-minimal then sonoff-classic did the trick. THX

mskrohn, 2018-07-01:
> @littlebilly you saved my day.
Uploading sonoff-minimal then sonoff-classic did the trick. THX

This worked!  Thank you!

tifrei, 2018-07-30:
Hi @ all,

I have the same error and cant upgrade my sonoff T1 2CH. 

Flashed with DOUT (1M 65K Spifs) 

When i try to upload the minimal version it still said Upload buffer miscompare. 

Is there any other solution or am i doing something wrong? is #define use_minimal a command i have to put in the console before uploading?

Thx in advance