arendst/Sonoff-MQTT-OTA-Arduino #158: Get only state of GPIO

Nayar, 2016-12-26:
Hi,

I am connecting a `HC-SR501` to my sonoff. However I wish to send the state only via MQTT so that my external server can decide what to do. e.g. it makes no sense to power on lights during the day when motion detected. I want my logics to be on the server-side.

The closest I've seen in the configs is `#define USE_WALL_SWITCH `. Is there an option that when activated, only sends something like `/stat/GPIO ON` or something. An option like `STAT` for `SWITCH_MODE` would be great.

arendst, 2016-12-26:
Define a ButtonTopic and you'll receive a cmnd/<ButtonTopic>/POWER message.

arendst, 2016-12-27:
See #141 too

closed 2016-12-27 by arendst

Nayar, 2016-12-28:
Unfortunately it changes the topic of the physical button of the sonoff itself. I'll try to see what can be done