arendst/Sonoff-MQTT-OTA-Arduino #164: Infallibility

paveleremin, 2016-12-28:
Hello everyone,

Yesterday I've got my first Sonoff TH and I decide to test own firmware before change it to Sonoff-MQTT-OTA-Arduino.

1) I setup automatic mode with minTemp and maxTemp - all worked fine
2) Then I turned off the router, and it's continue to work!

The question is, will Sonoff TH continue work without Wifi and/or Internet?

If not, I think it's not safe use it with gas boiler, because in this case we have two points of failure: router and mqtt server

arendst, 2016-12-28:
With my software it will continue to work as long as you configured it correctly:
- Once wifi is set-up functional change wificonfig to restart (```wificonfig 4```).
This way it will alternate between both configured AP's looking for a connection without a reboot and reconnect as soon as one of the AP's is online again.

paveleremin, 2016-12-28:
@arendst if I understood you correctly: Sonoff will try find and connect to wifi. But gas boiler will be ON or OFF, because all logic stored at, for example: OpenHAB or some MQTT service.
Once again ... with the standard firmware boiler will continue to work according to the set temperature.

Let's imagine Gas boiler + Sonoff TH with  Sonoff-MQTT-OTA-Arduino that used as thermostat:
1) Temperature range was setuped correctly and it's work fine
2) @arendst what will be with boiler if wifi is disappeared (AP was switched off)?

arendst, 2016-12-28:
Nothing will happen; Sonoff TH will try to connect to one of two AP's but fail. Relay will stay in same state as before and when temperature falls below your threshold nothing will happen too as Sonoff TH has no wifi connection and therefor will not receive info from OpenHAB.

closed 2016-12-29 by arendst